sfstar/hass-victron #81: After updating HA to 2023.6.0 I am getting error messages

gb095666, 2023-06-07:
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/victron/number.py:154
Integration: victron (documentation, issues)
First occurred: 23:39:17 (1 occurrences)
Last logged: 23:39:17

Unexpected exception importing platform custom_components.victron.number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 833, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 850, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/victron/number.py", line 154, in <module>
    @dataclass
     ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1223, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1213, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "/usr/local/lib/python3.11/dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'key' follows default argument

peregrines2, 2023-06-07:
Here too :( 
It seems to directly be related to Homeassistant 6.0.

[image]

[image]

peregrines2, 2023-06-07:
[image]

Dreanaught, 2023-06-08:
+1
I get the same error

andrewdavidson64, 2023-06-08:
Same problem here, typically I'd just upgraded my Cerbo to 3.0.  After downgrading and getting nowhere, I realise HA had helpfully upgraded itself and broken everything.  Again.

sfstar, 2023-06-08:
Thanks for reporting,

After this week I will have some time to fix this.
For now please revert to HA core 2023.05 and wait with upgrading until a patch release has been made.

gb095666, 2023-06-08:
> Thanks for reporting,
> 
> After this week I will have some time to fix this. For now please revert to HA core 2023.05 and wait with upgrading until a patch release has been made.

Thanks for replying, I have downgraded HA and its back working now, I will await your update when you get around to it.

Also, I am not sure if you have seen I have created a Pull request, apologies if I have done it wrong as I am still learning GIT etc.

Its an update for the Victron EV Charger section, I would be grateful if that can be included in the upgrade if possible.

sfstar, 2023-06-10:
V0.1.1 has been released to adress this issue.
Please let me know if any other issues arrive with this new release and ha core 2023.6.0

@gb095666 I've seen the PR and will address it in a coming release.
I've several specsheet mismatches issues open with victron at the moment.
When they respond on what the correct definition for these (and your evcharger decoding issue) I will be able to proceed.

peregrines2, 2023-06-10:
THANK YOU for all your hard work and for updating so quick!

LeoWinterDE, 2023-06-10:
Thanks for the fast update, error is gone.

gb095666, 2023-06-10:
> V0.1.1 has been released to adress this issue. Please let me know if any other issues arrive with this new release and ha core 2023.6.0
>
> @gb095666 I've seen the PR and will address it in a coming [...]

@sfstar Thanks, as long as you have seen it that's fine.

sfstar, 2023-06-16:
No further issues were reported with ha 2023.06 and the v0.1.1 release.
Therefore closing this issue.
Feel free to re-open if the issue persists

closed 2023-06-16 by sfstar