sfstar/hass-victron #82: Cerbo 3.0 Update

Linusten, 2023-06-09:
Has anyone updated yet to 3.0 an can confirm the addon is still working? 
I am a little afraid to update yet :)

https://www.victronenergy.com/blog/2023/05/31/venus-os-v3-00-generator-start-stop-improvements-ess-peak-shaving/

[Venus OS - firmware changelog.docx](https://github.com/sfstar/hass-victron/files/11705290/Venus.OS.-.firmware.changelog.docx)

arturojimenezr, 2023-06-09:
Yo lo tengo a v3.0 y sigue funcionando.

Saarlothri, 2023-06-10:
Auch ich habe keine Probleme mit der Version 3.00

Linusten, 2023-06-10:
I have also upgraded now to 3.0, everything works fine :) thank you all

closed 2023-06-10 by Linusten

sfstar, 2023-06-11:
Thanks for this feedback,

I will also update to v3 and update the readme to reflect te compatibility after a few days of testing