sfstar/hass-victron #83: States not updating

Tidle999, 2023-06-11:
### hello,

I just installed the integration and i love how easy it works without dealing with alle the modbus and mqtt stuff.
But for me the states (for example my solar chargers) are updating very infrequently. For example charger 1 last state was 10 min ago, charger 2 was 25min ago. When i am changing on the VRM portal it’s updating, so there is no issue in general. When i reload the integration its also updating. I changed the interval from 30s to 1s but no changes. Any ideas?
Are there any plans for integrate switches, for example to switch on/off the inverter?

braamvh, 2023-06-11:
Depends on the entity. If the value doesn't change (e.g. MPTT state), it will show as updated when the last state change happened. Check entities that do change, e.g. MPPT yield, and they will change frequently.

randyoo, 2023-06-13:
> When i reload the integration its also updating. I changed the interval from 30s to 1s but no changes. Any ideas?

This seems to be a known bug at the moment. I used [this workaround](https://github.com/sfstar/hass-victron/issues/21#issuecomment-1399328455), and it gets the job done for me, at least until the bug gets resolved. Meanwhile, 1s seems very aggressive. The documentation says "For slower systems setting the interval to lower than 5 seconds might cause issues."

sfstar, 2023-06-13:
I'm currently working on improved connection loss handling. (In the current release it's almost non-existent)
I hope to be able to release this in a relatively short timeframe since this is a major bug when it comes to quality of life of using the integration.
It will however take some time and some help from the community (the improved logic will be released as a beta once I believe that it should address all connection loss issues) to test the new logic before making it a stable release.
This is especially important since (allthough I am able to reproduce some of the reported connection stability issues) for some of the issues I have no clue on how to reproduce them.

randyoo, 2023-06-13:
> I'm currently working on improved connection loss handling.

@sfstar --
In my case, it doesn't seem to be a complete connection loss. What I've observed is that sometimes one (or more) entities become unavailable, as the rest continue to update. There doesn't seem to be any rhyme or reason to when an entity stops updating, though it seems to happen daily, at minimum. Reloading the integration brings it back every time, and since I've got that automated now, it's not a big deal for me.

Although I don't know how to reproduce the drops, either, I'm happy to help with the testing when the time comes.

pos-ei-don, 2023-06-14:
I see the same. Sometimes, a Single value stops getting Updates.

sfstar, 2023-06-20:
> @sfstar -- In my case, it doesn't seem to be a complete connection loss. What I've observed is that sometimes one (or more) entities become unavailable, as the rest continue to update. There doesn't [...]

I do believe your issue is related to connection stability / read stability of the modbus server.
In the current release a device stops updating if a connection interrupt occurs after which the device won't be updated until a reload of the integration or a reboot of ha.
For those experiencing intermittent connections or values that stop updating you could try the code on the branch of this PR.
https://github.com/sfstar/hass-victron/pull/90

It partially addresses improved connection handling by using the coordinator push mechanism for all entities instead of polling from the entity side every <interval> regardless if there is new data available.
I believe that mixing these two mechanism I caused the no longer updating behaviour and some of the connection issues.
Please note that although the code on that branch is stable enough that i'm running it on my production HA instance (without issues since I switched over) it isn't yet releasable since I need to improve the logging around the connection handling and availability in cases like when the entire GX device goes down.

randyoo, 2023-06-23:
>  For those experiencing intermittent connections or values that stop updating you could try the code on the branch of this PR. #90 

@sfstar -- I do apologize, but I just reviewed the automation I set up to restart your integration automatically whenever entities stopped updating. It hasn't triggered in 10 days! I do believe that is about how long it's been since I upgraded the VenusOS to 3.00, so that's most likely what "fixed" it for me--which means I'm not able to test your PR myself.

sfstar, 2024-05-19:
Several improvements regarding connection stability were made since this issue was reported.
Are the state update issues still occuring (while running the v0.3.0 release)?
If so, please provide debug logging for the integration in this thread.
The issue will be closed if no reply is received before 19 august 2024

sfstar, 2024-08-20:
Closing issue as stale.
Feel free to open a new issue if the issue persists.

closed 2024-08-20 by sfstar