sfstar/hass-victron #88: victron_maxchargecurrent stopped working since last update

pos-ei-don, 2023-06-20:
It used to work, but stopped in the current version!
I need it to see the difference between the maximum current allowed bx the battery and "what is left" after the DC solar chargers.

See the Plot. I tried redetection and reloaded the plugin several times.

[image]

sfstar, 2023-06-20:
What does the debug logging say?

pos-ei-don, 2023-06-22:
It does not contain anything when searched for "maxchargecurrent"

The only entry regarding victron I found is
"Error fetching victron data: Fetching registers failed"
.
But most other Entities still work as bevore. I thought maybe a change in the last update of this integration ?!?
In the code I only found this line, and it didn't get a change for a while, so I think I cannot help here.
https://github.com/sfstar/hass-victron/blob/7517db588b601a958754b6f9dcd3f0310588ce35/custom_components/victron/const.py#L603

trentjw, 2023-06-29:
I have to go in and reload it on almost a daily bases in order for it to detect any changes in the charging system

pos-ei-don, 2023-07-30:
I just found out that it depends on the configured BMS, if this value is available or not.
I just Set my Main BMS to maintenance and switched to a different one, and voila, this value is back again!

sfstar, 2024-03-14:
Closing issue as solution was found outside of this integration.
Feel free to re-open if you believe this issue originates from within the integration

closed 2024-03-14 by sfstar