sfstar/hass-victron #92: L3 usage not shown correctly

AminShAT, 2023-06-26:
Hi, I compared the usages shown in HA with the ones in VRM and on the Cerbo and for me, L3 is missing about 600W in the sensor sensor.victron_vebus_out_l3_power_227

Where Cerbo and VRM are showing 2.100 Watt, HA is only showing 1.500 Watt right now.

pos-ei-don, 2023-06-28:
I checked. Cannot see this here.

AminShAT, 2023-06-28:
It seems to be different values, but right now, I again see a difference between VRM and HA:

AminShAT, 2023-07-07:
Consumptions are still not correct and do not sum up to battery in/out and grid in HA. Victron VRM shows different consumption s well.

sfstar, 2024-03-29:
Hello AminShAT,

Perhaps there is an difference between the polling of the integration and perhaps vrm averaging consumption over a larger period of time.
Is the load consistant during the entire period and is the delta exactly the same during the period in question?

AminShAT, 2024-04-10:
I found the reason:

The consumption parameters do not take into account AC PV production. As I had so far all of my microinvertes on L3, it only affected L3.

In order to calculate the real consumption one needs to substract system.pvoutput (at least in my case).

Issue can be closed.

sfstar, 2024-04-12:
Thank you for reporting back.
Closing this issue as informed by reporter.

closed 2024-04-12 by sfstar