sfstar/hass-victron #96: VECan Device is not availabe after HA restart

dersch81, 2023-07-05:
Using v 0.1.2

After a HA restart all the victron devices are up. But one VEcan device (the only one in the system) is unavailable. I need to manually reconfigure => rescan devices to get it back available in HA.

dummy74, 2023-07-05:
I have the same problem. After restarting HA I always have to reconfigure the victron integration and set the rescan option. 
This problem has existed since the beginning.

sfstar, 2023-07-05:
Thank you for opening this issue.
Does the vecan modbus id change between reboots?
What does the logging say, which versions are you running (gx device, this integration, ha)

dersch81, 2023-07-07:
Hi no, the vecan modbus id ist always 0 

I'm running Venus 3.0, Integration 0.1.2 and HA 2023.6.3

ppbering, 2023-10-07:
Hello,
I have the same issu on my system, every time HA restart (for any update the system as to deal with), I MUST rescan available devices to be able to retrieve values from one of my solar array.
And I have not been able ton understand what is happenning.
I only loose one device, but it's making a huge mess in all my automations and the energy dashboard.

Thanks for your time on this matter.

Regards

dersch81, 2023-11-15:
is there no way to just rescan automatically after restart of HA?

viccooper142, 2024-01-14:
Same issue. Is there a file we can edit to make rescan happen after every restart? Or make it so the check box under configuration stays checked?

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar