sfstar/hass-victron #97: Setup Failed after installing Homeassistant 2023.7.1

peregrines2, 2023-07-07:
Installed new Homeasssistant July 7.1 Update.
Receiving the following failure message:

[image]

BeardOnTheRange, 2023-07-07:
I found after a reboot it seemed to come back online

peregrines2, 2023-07-07:
Thank You very much! A restart did - indeed - bring it back to life :)

sfstar, 2023-07-07:
The new core version of ha most likely uses a newer version of the pymodbus library that was overriden with the version for this integration upon the reboot

detlefh68, 2023-07-18:
But only every 2nd new start. When I reboot HA, I get the message below. After the second new start of HA the integrations comes back.

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: homeassistant.setup
Source: custom_components/victron/coordinator.py:13
Integration: victron (documentation, issues)
First occurred: 12:45:20 (1 occurrences)
Last logged: 12:45:20

Setup failed for custom integration victron: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 13, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)

sfstar, 2023-09-04:
Please test the beta v0.1.4 release of this interation (this is a copy paste reply which isn't customized for each issue v.0.1.4 is attempting to adres).
Make sure that (in cae this applies to you) the core modbus integration is also fully enabled after bumping to version 0.1.4 and let me know if the issue persists.

sfstar, 2024-03-15:
Closing issue as resolved.
Feel free to re-open if the issue persists

closed 2024-03-15 by sfstar