sfstar/hass-victron #98: No device found - is Venus OS V3 still not supported?

MB1983k, 2023-07-07:
I do have Venus OS V3 running on a Raspberry. It is possible to install it in Home Assistant and it seems all fine - but there are no devices...

I had to upgrade to V3 now as I had some issues with the Victron system before... Before this upgrade it was running well...

andlo, 2023-07-07:
It works for me just as good as venusos v2
All devices is found, but need to rescan after restart of Homeassistant as GPS seems to have wrong id's until a rescan. Same "issue" as on venusos v2 though.

MB1983k, 2023-07-08:
Ok… interesting… It seems to be absolutely not working anymore for me…

sfstar, 2024-03-15:
v0.1.7 contains the fix for the GPS unit of measurement.
v3.0 should be supported by the integration as of that version as well (running v3.20 at my end without issues).
The only thing to note is that there are newly added registers missing which will be included in this integration starting when #174 is landed

sfstar, 2024-05-19:
Closing issue as answered / validated by running v3.x without issues for some time now.
Feel free to re-open if you believe I missed something

closed 2024-05-19 by sfstar