sfstar/hass-victron #100: ESS AC POWERSETPOINT can't be set

AminShAT, 2023-07-13:
This field is used to engage battery load from the grid. The integration shows changes in the value correctly, but it's value can't be set from within the home assistant.

AminShAT, 2023-07-13:
Addition:

The values are not displayed in the entities when changed from outside, they are just logged correctly.

LeoWinterDE, 2023-08-14:
Have you tried deleting once and adding the victron connection again? Then it worked for me.

sfstar, 2024-03-15:
Hello @AminShAT and thank you for opening this issue,

Could you perhaps provide me with a screenshot of your hass-victrons integration configuration?
This would help me in narrowing down the cause of your issue since the values provided in the integrations configuration are used to calculate safe min/max values for entities.

AminShAT, 2024-03-15:
Du you mean this one?

[image]

sfstar, 2024-03-17:
Exactly.
In your configuration you specify an AC per phase current limit of 0.
Since this is the base for calculating the powersetpoint range (namely grid V * grid A) this results in the behaviour you are experiencing.
By setting the current limit to an value in AMPS you will see the slider entity become useable for setting the powersetpoint.

AminShAT, 2024-03-17:
I understand, but this is inconsistent with the behavior in Victron itself and from outside via Nodered, where it is possible. Not sure, if it would be best practice, to have 2 places to maintain system characteristics. Aren't those values available within Victron?

sfstar, 2024-03-18:
Unfortunatly the possibility of discovering settings in the GX device is limited via the modbus server.
Therefore, a register scan for all valid unit id's was implemented and amps/voltages are explicitly asked for.
Since, the integration cannot retrieve the values from the GX device.

AminShAT, 2024-03-18:
Alright. So set it to 65A. Will try next time. I used actually to force load my batteries.

sfstar, 2024-03-29:
Great to hear.
I'm closing the issue as resolved.

closed 2024-03-29 by sfstar