sfstar/hass-victron #102: Error after upgrading to v0.1.2

marnitz, 2023-07-23:
After upgrading from v0.1.1 to v0.1.2 this morning none of my entities are updating. The following error shows in the logs:

```
Logger: homeassistant
Source: custom_components/victron/sensor.py:157
Integration: victron ([documentation](https://github.com/sfstar/hass-victron), [issues](https://github.com/sfstar/hass-victron/issues))
First occurred: 8:25:55 AM (30 occurrences)
Last logged: 8:28:20 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 229, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 374, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 164, in async_update_listeners
    update_callback()
  File "/config/custom_components/victron/sensor.py", line 157, in _handle_coordinator_update
    self._attr_native_value = self.entity_type.decodeEnum(data).name.split("_DUPLICATE")[0]
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 711, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 1128, in __new__
    raise ve_exc
ValueError: 5.0 is not a valid generic_status
```

Restarting, removing and re-adding the hub, removing and re-downloading the extension does not resolve the issue. Downgrading to 0.1.1 is also does not resolve the problem.

Home Assistant 2023.7.2

marnitz, 2023-07-23:
I've managed to get this working again. 
Currently running HA 2023,7,3 and hass-victron v0.1.2. I have had to disable the "Tank Status"
[image]
I assume it's the currently disconnected tank that is causing the problem:
[image]

sfstar, 2024-03-15:
Hello @marnitz and thank you for opening this issue,

Would you be able to re-enable the tank status and check what the status in the GX device is when the above mentioned error is returned?
It seems that there are possible states for this register that are undocumented by victron.
Thereby, decoding of the register fails and the error is logged.

By checking the state in the GX device when the error occurs we can deduce what the state of register value 5 is supposed to be.

marnitz, 2024-03-17:
@sfstar - the status is "Error"

[image]

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar