sfstar/hass-victron #104: pv inverter: Missing values in home assisant

alex1891, 2023-07-27:
Hi, 

I am using this plugin to read out my kostal pv inverter: https://github.com/schenlap/venus_kostal_pico and it is doing well.
[image]

When I am using your great home assisant plugin, everything is working fine exepct one missing (and very important) value: The total generated energy is missing in home assisant:
[image]

Can you please help me with that?

Best regards
Alex

bifbaf, 2023-10-18:
Has someone a solution for this? I am also missing the total amount of the kwh that has collected from the inverter so that I can use this entity for the energy dashboard.

alex1891, 2023-10-18:
I do not have a clean solution for that. My (very dirty) workarround: I have changed the python code in my cerbo GX - I write the value of "total energy forward tota"l on the variable "L1 energy forward total". Not nice, but it works for know (I don´t need the values of the single phases)

bifbaf, 2023-10-18:
So the Sum of 

L1_energy_forward_total
L2_energy_forward_total
L3_energy_forward_total

Is the correct value that I can use for the dashboard? If this is the case I can create a new value with a template out of these entities.

closed 2025-03-05 by github-actions[bot]