sfstar/hass-victron #105: Intégration failed to load, setup since update HA 2023.8.0

dbachats, 2023-08-03:
Dear sfstar hi and sorry to disturb you at this summer time.
Since updating HA to version 2023.8.0 the integration is failing to setup (despite  multiple rebooting)
Thank you again for the time and work

pos-ei-don, 2023-08-03:
it works here, so maybe there is something special in you setup?

dbachats, 2023-08-03:
Thanks for the reply i will try to reinstall the intégration

dbachats, 2023-08-03:
If this can help someone.
I first reboot the gx device through the remote console and everything got working good
Thanks 🙏

Skycomm, 2023-08-06:
I've encountered the same issue since updating to version 2023.8.0, particularly with the modbus direct entries, which have ceased to function properly.

Upon testing, I discovered that the validate config function failed, and a reset attempt resulted in an error. Interestingly, commenting out anything related to modbus in the config file restored normal function, and the Victron system resumed working. It appears that something might have changed with modbus in this update.

However, I've faced a recurring issue with the Victron app on reboot: the GX works, but the VE.Can inverter doesn't. If I rescan, the inverter works, but I lose the GX functionality. This alternating failure between the two systems is perplexing.

I'm not sure if this detailed insight will assist anyone else facing these problems, but I wanted to contribute this information to the community. Any feedback or additional insights would be appreciated.

dbachats, 2023-08-06:
Thanks for the info. Since the last GX device reboot everything is working well. Sometimes mostly after an HA restart the victron integration is failing to load leaving sensors unavailable. If so, i firstly restart HA again and most of the times every victron sensor reloads correctly. If not i reboot the gx. I Don't know why it is behaving like this. Hope helps someone

Skycomm, 2023-08-07:
ill try that tonight.. the victron plugin is working ill just add to anyone else readying the mudbus issues only started since the start of 23.8 wasn't present in the betas.

thanks for the extra info all

sfstar, 2023-09-04:
Please test the beta v0.1.4 release of this interation (this is a copy paste reply which isn't customized for each issue v.0.1.4 is attempting to adres).
Make sure that (in cae this applies to you) the core modbus integration is also fully enabled after bumping to version 0.1.4 and let me know if the issue persists.