sfstar/hass-victron #106: Seperate Entities for Feed in

BeardOnTheRange, 2023-08-04:
Perhaps I have a gap in my knowledge here, but it would be great to separate out entities for one's grid/battery to show the draw from, and the feedback separately. 

I.e. 100 Kwh fed back into the grid, and 120 Kwh used from the grid, rather than having a value that is positive & negative. 

If there is another way fo achieving this please enlighten me

cybertza, 2023-08-14:
You could probebly use a template for that if there is no planned changes, 
would be something along the lines of what i do for the battery
```
- platform: template
  sensors:
    battery_charge_watts:
      friendly_name: Battery Charge Watts
      unit_of_measurement: "W"
      value_template: >
        {% if states.sensor.victron_system_battery_power.state | int >= 0 %}
            {{ states.sensor.victron_system_battery_power.state | int }}
            {% else %}
            {{ '0' }}
            {% endif %}
            
- platform: template
  sensors:
    battery_discharge_watts:
      friendly_name: Battery Discharge Watts
      unit_of_measurement: "W"
      value_template: >
        {% if states.sensor.victron_system_battery_power.state | int <= 0 %}
            {{ -1* states.sensor.victron_system_battery_power.state |int(default = 0)}}
            {% else %}
            {{ '0' }}
            {% endif %}
```

BeardOnTheRange, 2023-08-14:
Thank I will give that a try

mmaritz, 2023-08-23:
I have tried this and the integration warns of any victron .yaml in configuration.yaml will be ignored.

I am keen to get my battery sensor data into HA but think it's better for the integration to measure the battery energy flow and give the input and output data.  Can this be added to the Dev list please.

BeardOnTheRange, 2023-08-24:
Thanks to @cybertza I used the below code to achieve what I wanted. @mmaritz I think the same will work for you:

So I used the Victron integration, and then the below code to separate consumption & production. Noting that this will give you a  power figure, not energy. I then used a Riemann Summ (left) helper to convert from power to energy. That way you should have all the data you need.

- platform: template
    sensors:
      grid_consumption:
        friendly_name: "Grid Consumption"
        unit_of_measurement: "W"
        value_template: >-
          {% set grid_power = states('sensor.victron_system_grid_l1')|float %}
          {% if grid_power > 0 %}
            {{ grid_power }}
          {% else %}
            0
          {% endif %}
      grid_feed_in:
        friendly_name: "Grid Feed In"
        unit_of_measurement: "W"
        value_template: >-
          {% set grid_power = states('sensor.victron_system_grid_l1')|float %}
          {% if grid_power < 0 %}
            {{ -grid_power }}
          {% else %}
            0
          {% endif %}
      battery_discharge_power:
        friendly_name: "Battery Discharge power"
        unit_of_measurement: "W"
        value_template: >-
          {% set battery_power = states('sensor.victron_system_battery_power')|float %}
          {% if battery_power > 0 %}
            {{ battery_power }}
          {% else %}
            0
          {% endif %}
      battery_charge_power:
        friendly_name: "Battery Charge Power"
        unit_of_measurement: "W"
        value_template: >-
          {% set battery_power = states('sensor.victron_system_battery_power')|float %}
          {% if battery_power < 0 %}
            {{ -battery_power }}
          {% else %}
            0
          {% endif %}

cybertza, 2023-08-24:
@BeardOnTheRange can you share your w->kwh /kwhd config as well please. That helper confuses me

BeardOnTheRange, 2023-08-24:
I didn't have need to change it on the sensors themselves, I use the Powerflow plus and Energy Flow plus cards, and they have an inbuilt conversion. Those are pretty much the only places I view the info so that was adequate for me.

https://github.com/flixlix/energy-flow-card-plus

https://github.com/ulic75/power-flow-card

cybertza, 2023-08-24:
> I then used a Riemann Summ (left) helper to convert from power to energy.

This part ;p

BeardOnTheRange, 2023-08-24:
ah right, so you can tell if it's a power sensor if it's not cumulative data, an example in my case from the code I posted earlier, it would be sensor.grid_feed_in

You would then create a new helper like below, and call it something like Grid_feed_in_energy
[image]

mmaritz, 2023-08-25:
I'm struggling and would appreciate an eye cast over the code below... works in template and no errors when I check the .yaml.
I have also tried to create the integrations using the helper.  

`victron:  
  sensors:
    - platform: template
      victron_battery_power_input:
      device_class: power
      state_class: measurement
      friendly_name: "Battery Input Power"
      unit_of_measurement: "W"
      value_template: >
        {% if states('sensor.victron_system_battery_power')|float(0) >= 0 %}
        {{ states('sensor.victron_system_battery_power')|float(0) }}
        {% else %}
        0
        {% endif %}
      availability_template: >
            {{ states('sensor.victron_system_battery_power')|is_number}}

    - platform: template
      victron_battery_power_output:
      device_class: power
      state_class: measurement
      friendly_name: "Battery Output Power"
      unit_of_measurement: "W"
      value_template: >
        {% if states('sensor.victron_system_battery_power')|float(0) < 0 %}
        {{ states('sensor.victron_system_battery_power')|float(0) }}
        {% else %}
        0
        {% endif %}
      availability_template: >
        {{ states('sensor.victron_system_battery_power')|is_number}}

    - platform: integration
      source: sensor.victron_battery_power_output
      name: Victron Battery Output Riemann
      method: left
      unit_prefix: k
      round: 3

    - platform: integration
      source: sensor.victron_battery_power_input
      name: Victron Battery Input Riemann
      method: left
      unit_prefix: k
      round: 3
`

cybertza, 2023-08-25:
Yea I used the visual helper, and it seems to have worked. [image]

mmaritz, 2023-08-27:
> I'm struggling and would appreciate an eye cast over the code below... works in template and no errors when I check the .yaml.
> I have also tried to create the integrations using the helper.

Solved: I have solved this via a Node-Red flow so my issue can be ignored if needed.

bjk201, 2023-12-17:
> > I'm struggling and would appreciate an eye cast over the code below... works in template and no errors when I check the .yaml.
> 
> > I have also tried to create the integrations using the helper.
> 
> 
> 
> Solved: I have solved this via a Node-Red flow so my issue can be ignored if needed.

Can you Share your flow?

closed 2025-03-05 by github-actions[bot]