sfstar/hass-victron #107: dont work for me

juanmijm, 2023-08-11:
I can't configure the integration, when I click on add integration I get this message:

[image]

I have installed version 0.1.2 from hacs and I have home assistant in the latest version, 2023.8.1

any solution? thank you!

cybertza, 2023-08-14:
full reboot HASS

juanmijm, 2023-08-15:
> full reboot HASS

I have restarted HASS and the server several times, and it still doesn't work

juanmijm, 2023-08-24:
> full reboot HASS

Today I have seen the new update of the integration, excited I have updated, restarted the server, and I have been disappointed again to verify that the same error continues... :(

cybertza, 2023-08-25:
No problem on my side ?Regards

juanmijm, 2023-08-25:
I don't know where the problem is, but this is the only thing that doesn't work in my Home Assistant and I'm annoyed.

If I configure the modbus connection directly in yaml code it works for me, but it's cumbersome and I'd like to use your integration.

juanmijm, 2023-08-25:
[image]

cybertza, 2023-08-25:
https://github.com/home-assistant/core/issues/65785

so looking at some other posts it seems that one of your other integrations could be causing the issues, there seems to be hints at how to trouble shoot this, its doesnt seem to be related to this plugin but infact a plugin and therefor home assistant related.

Analizing the logs like they have perhaps give us insight, i assume it would be with pymodbus or something, but its hard to guess, as im not sure what all gets loaded with this plugin.

juanmijm, 2023-08-25:
I have another Ingeteam brand inverter, also configured by modbus with another HACS integration, surely it could be due to that? so I can't have two integrations that use modbus at the same time?

juanmijm, 2023-08-28:
https://github.com/vortizhe/home-assistant-ingeteam-modbus

This is the another integration, thanks!

sfstar, 2023-09-04:
Based on a quick look at the provided link I saw that pymodbus version 2 was still used by that integration.
Which is no longer compatible with this integration and the ha core mdobus (yaml) integration. (both are using a >v3.1.0 release for quite some time now). 
If you want this integration to work it is best to either disable the integration that is using the older pymodbus version or to get them to update to the v3 library version since it's also not ensured that the yaml modbus integration remains working with a <v3.3.1. library version

juanmijm, 2023-09-05:
Thank you very much, I will tell the creator of the other integration

juanmijm, 2023-09-12:
I forgot to close the issue, thanks again!

closed 2023-09-12 by juanmijm