sfstar/hass-victron #113: Victron integration stops updating values

iJAY86, 2023-08-27:
After updating my the victron integration, I noticed it stops updating/giving current values after a while and only starts giving the current values after reloading the integration.

shaman171, 2023-08-27:
I have exactly same behaviour.
Latest update 0.1.3
Home Assistant 2023.8.4
Supervisor 2023.08.1
Operating System 10.5
Frontend 20230802.1 - latest

No changes in meantime on the Victron settings or equipment.

askpatrickw, 2023-08-28:
@shaman171 and @iJAY86, are you seeing errors like this?

[image]

If so, for which devices? My Battery and my Multiplus-ii seem fine, but my Lynx Smart BMS has these errors.

shaman171, 2023-08-28:
No I do not see any of such errors in HA. There are just no longer updated data and it stop on some "random" last value which was readed from Victron. 
Then HA simply "think" that thi value is right and it get no longer any more value and show this one. Once you let this Victron add-on start again it again get values for like 5-10 minut and then get dead again.

iJAY86, 2023-08-28:
> @shaman171 and @iJAY86, are you seeing errors like this?
> 
> [image]
> 
> 
> If so, for which devices? My Battery and my Multiplus-ii seem fine, but my Lynx Smart BMS has these errors.

No errors

askpatrickw, 2023-08-28:
I wonder if we're all seeing this issue: #83

trooster, 2023-08-28:
Same issue here.., data no longer updating. And I also had the solar data entities removed, simular to what 'askpatrickw' is experiencing.

jwiesel, 2023-08-28:
Same issue here. The integration worked nicely on version 0.12. Since I upgraded a few days ago it hangs after a short while messing up my PV-based EV-charging.

Log entries (both occured at the time the integration stopped updating data):

Warning:

> Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht (Error caused by a custom integration)
> 
> Logger: custom_components.victron.coordinator
> Source: custom_components/victron/coordinator.py:76
> Integration: victron (documentation, issues)
> First occurred: 26. August 2023 um 18:34:29 (1 occurrences)
> Last logged: 26. August 2023 um 18:34:29
> 
> no valid data returned for entities of slave: 20 (if the device continues to no longer update) check if the device was physically removed. Before opening an issue please force a rescan to attempt to resolve this issue

Error:

> 
> Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht (Error caused by a custom integration)
> 
> Logger: custom_components.victron.coordinator
> Source: helpers/update_coordinator.py:229
> Integration: victron (documentation, issues)
> First occurred: 26. August 2023 um 18:34:29 (1 occurrences)
> Last logged: 26. August 2023 um 18:34:29
> 
> Error fetching victron data: Fetching registers failed

jwiesel, 2023-08-28:
Not sure, if it is related. The latest Home Assistant bugfix mentions something about Modbus Slave issues: 
https://www.home-assistant.io/blog/2023/08/02/release-20238/#release-202384---august-24
https://github.com/home-assistant/core/pull/98644. I'll upgrade to 2023.8.4 and monitor, if that helps.

trentjw, 2023-09-01:
same problem quits updating until I do a reload of the integration then it works for about a day

trooster, 2023-09-01:
The only change from version 0.1.2 to 0.1.3 was an update to the PyModbus library from version 3.1.1 to 3.4.1.
https://github.com/sfstar/hass-victron/commit/36727e8764538f77263d3e0d6fa1b9d46cfaaf80

Version 3.5.0 is already released, and it loos like it has several fixes with the word "retry" in the name.
https://pymodbus.readthedocs.io/en/stable/changelog.html

trentjw, 2023-09-01:
> The only change from version 0.1.2 to 0.1.3 was an update to the PyModbus library from version 3.1.1 to 3.4.1. [36727e8](https://github.com/sfstar/hass-victron/commit/36727e8764538f77263d3e0d6fa1b9d46cfaaf80)
> 
> Version 3.5.0 of the PyModbus library is already released, and it looks like it has several fixes with the word "retry" in the name. https://pymodbus.readthedocs.io/en/stable/changelog.html

okay still have the issue of the integration freezing and not responding so not sure what you are getting at?

trooster, 2023-09-01:
@trentjw What I am getting at is that you either need to dowgrade to version 0.1.2 or wait until the author releases a new version which contains the 3.5.0 library of PyModbus which hopefully fixes the issue we are having.

jwiesel, 2023-09-02:
> Not sure, if it is related. The latest Home Assistant bugfix mentions something about Modbus Slave issues: https://www.home-assistant.io/blog/2023/08/02/release-20238/#release-202384---august-24 [home-assistant/core#98644](https://github.com/home-assistant/core/pull/98644). I'll upgrade to 2023.8.4 and monitor, if that helps.

Installing Homeassistant 2023.8.4 unfortunately did not improve the situation. The Victron integration still hangs after some time

Klumpen58, 2023-09-04:
I have the exact same problem since the upgrade.  😟

trentjw, 2023-09-05:
reverting back to version 1.2 as suggested is a fix for now until it is broken later I went back to it and it reporting as it should 

https://github.com/sfstar/hass-victron/tree/v0.1.2

mploetner, 2023-09-11:
Same problem even with the latest update 0.1.5 

Stops updating data for all entities after short time

pos-ei-don, 2023-09-11:
I see this after a short network disconnect. The plugin does not restart after the network Returns.

I have a unreliable connection over LTE, so I see this Situation almost daily.

mploetner, 2023-09-11:
@pos-ei-don could be the reason. I will check this. Was never a problem for months until around JULY

shaman171, 2023-09-11:
I can confirm also for me after update of HA and victron to latest versions it works already since yesterday without troubles. 
Thank you for fix! 👍

mploetner, 2023-09-12:
Ok Internet connection was stable but updating stopped for all entities about 8h ago. Any other ideas to solve it?

pos-ei-don, 2023-09-12:
Just to be clear: I spoke from small interruptions, nothing you can find with a ping per second.

mploetner, 2023-09-12:
Ok that could be a reason. I also use LTE connection. 

I update my VenusOS device hoping this will solve the problem but no luck. It works for some hours and then stops updating.

sfstar, 2024-05-19:
Since this issue was reported there have been several improvements made regarding connection stability and recovery.
Are the issues you are experiencing still regularly occurring or can the issue be closed as resolved?

mploetner, 2024-05-19:
It happens occasionally every few weeks. Reloading the integration solves the problem.

naofireblade, 2024-06-10:
I found a solution for my update problem. I use the external control ess mode and set the acpowersetpoint every 10 seconds (that works fine). In order to update the sensor values im home assistant, the configured update interval for the plugin must be < 10 seconds (e.g. 9 seconds).

Reason and possible bug in the plugin:
Whenever i set acpowersetpoint the plugin prints `[custom_components.victron.coordinator] Manually updated victron data` in the debug log but does not update the sensor values in home assistant.
Now after configuring the update interval for 9 seconds, i see `custom_components.victron.coordinator] Fetching victron data` and `[custom_components.victron.coordinator] Finished fetching victron data in 0.463 seconds (success: True)`in the logs.

sfstar, 2024-07-02:
Sorry was on holiday the last couple of weeks and noticed remotely once that the integration indeed stops updating values. So this issue needs further investigation.

trentjw, 2025-02-04:
Has this been fixed

jwiesel, 2025-02-05:
for me it is running stable for several months now. No further issue from my side

closed 2025-06-04 by github-actions[bot]