sfstar/hass-victron #115: Component error: modbus

fuxe13, 2023-08-31:
Component error: modbus - cannot import name 'Defaults' from 'pymodbus.constants' (/usr/local/lib/python3.11/site-packages/pymodbus/constants.py)

finde keinen fehler, bzw komme nicht zu diesem verzeichnis

pos-ei-don, 2023-08-31:
Kommentiere in der configuration.yml die modbus Einträge aus dann geht's wieder. Ist natürlich maximal ein workaround.

fuxe13, 2023-09-01:
Habe ich gemacht, bzw. neue yaml angelegt..dann kam dieser fehler:

Invalid config for [influxdb]: [modbus] is an invalid option for [influxdb]. Check: influxdb->influxdb->modbus. (See /config/configuration.yaml, line 23).

aus der conig ab zeile 23:
influxdb:
  username: xxxx
  password: xxxx
  tags:
    source: HA
  tags_attributes:
    - friendly_name
  default_measurement: state
  exclude:
    entities:
      - zone.home
    domains:
      - persistent_notification
      - person
  include:
    domains:
      - sensor
      - binary_sensor
      - sun
      - light
      - cover
    entities:
      - weather.home

TonyLar, 2023-09-04:
I got the same problem. Modbus stopped working 2-3 weeks ago. As a workaround I have comment out all modbus code in configuration.yaml to be able to restart Home Assistant. So at the moment modbus is not working for me.

# modbus:
#   - name: hub1
#     type: tcp
#     host: 192.168.1.20
#     port: 502
#     close_comm_on_error: true
#     message_wait_milliseconds: 10000
#     delay: 20
#     timeout: 20
#     sensors:
      # SYSTEM
      # MPPT Solar Arch
    #   - name: "Solar Charge state" # \\ 0=Off;2=Fault;3=Bulk;4=Absorption;5=Float;6=Storage;7=Equalize;11=Other (Hub-1);252=External control
    #     scale: 1
    #     precision: 1
    #     slave: 239
    #     address: 775
    #   - name: "Solar Voltage"
    #     data_type: uint16
    #     unit_of_measurement: "V"
    #     slave: 239
    #     address: 776
    #     scale: 0.01
    #     device_class: voltage
    #   - name: "Solar Current"
    #     data_type: int16
    #     unit_of_measurement: "A"
    #     slave: 239
    #     address: 777
    #     scale: 0.1
    #     device_class: current
    #   - name: "Solar Power"
    #     data_type: uint16
    #     unit_of_measurement: "W"
    #     slave: 239
    #     address: 789
    #     scale: 0.1
    #     device_class: power
    #     state_class: measurement
    #   - name: "Solar Operation mode"
    #     data_type: uint16
    #     slave: 239
    #     address: 791
    #     scale: 1
    #   - name: "Solar Yield garage today"
    #     data_type: uint16
    #     unit_of_measurement: "kWh"
    #     slave: 239
    #     address: 784
    #     scale: 0.1
    #     precision: 3
    #     device_class: power
    #     scan_interval: 20
    #   - name: "Solar Max power today"
    #     data_type: uint16
    #     unit_of_measurement: "W"
    #     slave: 239
    #     address: 785
    #     scale: 1
    #     device_class: power
    #   - name: "Solar Yield garage yesterday"
    #     data_type: uint16
    #     unit_of_measurement: "kWh"
    #     slave: 239
    #     address: 786
    #     scale: 0.1
    #     precision: 3
    #     device_class: power
    #     scan_interval: 20
    #   - name: "Solar Max power yesterday"
    #     data_type: uint16
    #     unit_of_measurement: "W"
    #     slave: 239
    #     address: 787
    #     scale: 1
    #     device_class: power
    #   - name: "Battery Voltage"  # // Battery Voltage determined from different measurements. In order of preference: BMV-voltage (V), Multi-DC-Voltage (CV), MPPT-DC-Voltage (ScV), Charger voltage //com.victronenergy.system
    #     unit_of_measurement: "V"
    #     scale: 0.1
    #     precision: 1
    #     slave: 100
    #     address: 840

sfstar, 2023-09-04:
Please test the beta v0.1.4 release of this interation (this is a copy paste reply which isn't customized for each issue v.0.1.4 is attempting to adres).
Make sure that (in cae this applies to you) the core modbus integration is also fully enabled after bumping to version 0.1.4 and let me know if the issue persists.

sfstar, 2023-09-05:
Thank you for testing @TonyLar.
To double check, the issue you are experiencing is still the 
```Component error: modbus - cannot import name 'Defaults' from 'pymodbus.constants' (/usr/local/lib/python3.11/site-packages/pymodbus/constants.py```?

If so what version of HA core are you running?
Have you upgrade home assistant core after upgrading this integration? (if so the suggestion at the end might not work for you)
Do you have any other modbus configuration in your instance with the core modbus integration?
Do you have any other custom integrations active that communicate over modbus (if so could you name them?)

What I think might have happened is that with hass-victron v0.1.3 an incompatible pymodbus library was installed that breaks your core modbus integration for the ha core version you are running.
Since HA has had multiple breaking version bumps for some of the pymodbus functionality in the releases since 2023.7 an specific core version could help narrow down the issue.
Having any other modbus components next to this integration and the core modbus integration could result in the suggestion at the end not working since that custom integration could enforce an older version of pymodbus. (in which case that integration should also be patched).

If you are able to access the ha console of your instance via either an attached monitor/keyboard and mouse (or via this addon https://community.home-assistant.io/t/home-assistant-community-add-on-ssh-web-terminal/33820)
Please execute the following command:

```ha core rebuild```

This will let homeassistant check if there were modifications to it's image after starting up.
If an integration (like the previous version of this integration originally did) installs an differing library than was provided by the base image it will result in the image being redownloaded and restarted.

This will ensure that an different pymodbus version that was installed by (the previous version of) this integration is wiped.
Please let me know if the issue persists after attempting the provided core command.

TonyLar, 2023-09-05:
Thanks for your quick response and excellent explanation. Here are the versions I'm running:
Home Assistant 2023.8.4 
Supervisor 2023.08.1 
Operating System 10.5 
Frontend-version: 20230802.1 - latest 

I've have two solarpanel installations that I've been monitoring without any problem for about a year. Since the Victron integration only can be configured to communicate with one Cerbo GX I have been using modbus to communicate with my second Cerbo GX. I have no other modbus configurations. I will try the ha core rebuild later today.

TonyLar, 2023-09-05:
Now I have executed ha core rebuild but unfortunately the issue persist. The only difference I see between v0.1.3 och v0.1.4 is when I enable "modbus:" in configuration.yaml:
With v0.1.3 I get the error message "Component error: modbus - cannot import name 'Defaults' from 'pymodbus.constants' (/usr/local/lib/python3.11/site-packages/pymodbus/constants.py"

With v.0.1.4 the code validation hangs (goes on forever) so there is no error message.

Maybe the different behavior can be a clue?

DigitalPals, 2023-09-05:
Looking forward to see a fix for this, I have the same issue. I reverted back to Home Assistant 2023.8.1 to fix this issue.

TonyLar, 2023-09-06:
Update: After update and total restart I got both modbus and the Victron integration running again. I'm using v0.1.4 together with:
Home Assistant 2023.8.4 
Supervisor 2023.08.3 
Operating System 10.5 Frontend-version: 20230802.1 - latest 

The validation of configuration.yaml also is ok.

Just been up for a few minutes now but I'll hope it will be stable now :-)

TonyLar, 2023-09-06:
Thanks for all support!

closed 2023-09-06 by sfstar