sfstar/hass-victron #118: Dynamic ESS

andzie82, 2023-09-04:
Possible to implement the new feature Dynamic ESS?
[image]
[image]

steve0564, 2023-09-07:
Where can I find further information about this feature?

sfstar, 2024-03-29:
Unfortunately I cannot implement what victron doesn't expose via the modbus server.
You could open an issue here: https://github.com/victronenergy/dbus_modbustcp/issues to request that they expose this information via an modbus register.
Until then I'm closing this issue as wontfix as the solution lies outside the scope of the integration (for now)

closed 2024-03-29 by sfstar