sfstar/hass-victron #121: Change host IP

cybertza, 2023-09-08:
I have a an interesting problem, i would like to EDIT the Provided IP for the host with the config, i have quite a bit of data, and i would like to specify, but do not want to add a new connector in fear of loosing historical data, and configuration data, Is there a flat file that can be edited for this somewhere or a way to edit the address, i just want to change it for EG
192.168.100.105
to 192.168.100.15

Thanks for anyone with insight.

flyrmyr, 2023-10-21:
Seconded. Switched to a new router and it moved from the `127.168.0.0` range to the `10.0.0.0` range. Spent hours renaming devices with this integration and setting up templates and automations, would love to avoid trying to re-do all that.

sfstar, 2024-02-25:
You could edit the .storage folder that contains the configuration entry for the configuration.
Otherwise removing and re-adding creates the same entities without wiping historical data.
Currently unsure if this is something that needs to be implemented in this integration. (Not seeing an advantage of removing and re-adding the hub at this time)

sfstar, 2024-04-06:
With Homeassistant version 2024.4 the following was introduced: https://developers.home-assistant.io/blog/2024/03/21/config-entry-reconfigure-step.
This could perhaps be used to provide the requested functionality

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar