sfstar/hass-victron #122: Integration broken after Update to 2023.9.1

peregrines2, 2023-09-09:
I updated to 2023.9.1 and the Integration is completely broken and wont start.
Restarting Venus OS or reinstalling the Integration does not help.

[image]

detlefh68, 2023-09-09:
I have the same here:

Logger: homeassistant.setup
Source: loader.py:826 
First occurred: 07:21:07 (1 occurrences) 
Last logged: 07:21:07

Setup failed for custom integration victron: Unable to import component: Exception importing custom_components.victron
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/victron/__init__.py", line 9, in <module>
    from .coordinator import victronEnergyDeviceUpdateCoordinator as Coordinator
  File "/config/custom_components/victron/coordinator.py", line 13, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 826, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.victron

sfstar, 2023-09-09:
Which version of the integration are you running?
Does this also happen with the beta release v0.1.4?

peregrines2, 2023-09-09:
I reverted back to v0.1.2 v0.1.3 and tried v0.1.4 beta.
No success. The integration is completely broken :(
I really rely on it unfortunately :(
Wish I had some skills to help with programming - unfortunately I cant :(

detlefh68, 2023-09-09:
I used the backup from version 2023.9.0 and the integration works again.

smout74, 2023-09-09:
There was a pymodbus upgrade in HASS 2023.9.1

[image]

mmaritz, 2023-09-09:
I had the same problem and downgraded HA back to 2023.9.0.

Suggestion:  A thread we can follow which notified us when the integration is upgraded to the latest version of HA or the highest version that is compatible.  We can then upgrade HA when we get notified of hass-victron being upgraded.

Thanks

sfstar, 2023-09-09:
The issue is indeed caused by the pymodbus bump in this homeassistant patch release. (Wierd that they break functionality in a patch release).
The newly referenced pymodbus library version has an change to the spelling of the endian.{Little, Big, Auto} constants for pymodbus.
In order to make this integration compatible with ha core 2023.9.1 It will entail a breaking change in this integration to.
I will make a release to address this issue

peregrines2, 2023-09-09:
Thank You for your work sfstar!

sfstar, 2023-09-09:
v0.1.5 has been released to address this issue.
Please ensure you update this integration to v0.1.5 and ha core to 2023.9.1 in the same patch round.

peregrines2, 2023-09-09:
Thank You very much for this quick fix!!!

peregrines2, 2023-09-09:
Already installed under 2023.9.1 and everything is working fine again!

gb095666, 2023-09-09:
Does 2023.9.1 include the fix for the EVcharger integration ?
Pull request # Update for Error from Victron EV Charger #68
Everytime I install a new update it seems to break the EVcharger integration and I have to manually add it again.

detlefh68, 2023-09-10:
Thank you very much for the fast fix!
2023.9.1 and v0.15 works fine 👍

tomsride, 2023-09-10:
Same here - works :-) 2023.9.1 and v0.15 👍

mmaritz, 2023-09-13:
> Suggestion: A thread we can follow which notified us when the integration is upgraded to the latest version of HA or the highest version that is compatible. We can then upgrade HA when we get notified of hass-victron being upgraded.

I have opened a discussion topic for this suggestion - hope this is the correct place ;)

sfstar, 2024-03-14:
Closing Issue as resolved, further tracking of @mmaritz his suggestion regarding notifications will be done in the discussion section

closed 2024-03-14 by sfstar