sfstar/hass-victron #125: PV yield power >~6550W starts from 0

cm2962, 2023-09-11:
Hi,
I see the PV yield power correctly up to about 6.550W.
E.g.
Actual Power / Shown Power in hass-victron
100 / 100
5.000 / 5.000
6.500 / 6.500
7.000 / 450 (=7.000 - 6.550)
10.000 / 3.450 (=10.000 - 6.550)

Am I making any wrong settings or is this a general issue?

Thanks Christian

PS. Excellent integration besides this :)

cm2962, 2024-02-25:
Just asking. Is there any news or potential help on this issue?

sfstar, 2024-03-14:
Hello @cm2962 and thank you for opening this issue,

Could you perhaps provide me with the entity names that stop at 6550 watt.
So that I may check if the register can only return values up to 6550 watt (and whether there are other registers available that report a larger range)

closed 2025-03-05 by github-actions[bot]