sfstar/hass-victron #128: Swapping an EM24 out with an EM540 - "grid" device not discovered properly again

andreas-bulling, 2023-09-24:
Hi,

I had an EM24 installied with my Victron system before I installed this integration. All went well - there was a "grid" component discovered and added to HA. Last week the EM24 was replaced with an EM540 - and now the "grid" device doesn't show anything anymore in HA, only blanks. Seems the new EM was not discovered by the integration. It shows properly in the Cerbo though...

pos-ei-don, 2023-09-24:
Goot to know, then I will postbone my change until this module is Adapted.
Can you give some impression, do you See a difference?

andreas-bulling, 2023-09-24:
What do you mean? In HA the difference is that I don't see any values from the grid meter anymore, which is rather annoying. The EM540 is obviously a lot faster and thus should allow the ESS system to adapt more quickly to changing loads, thus reducing residual consumption from the grid. I don't yet have numbers to back this up, though.

pos-ei-don, 2023-09-24:
> The EM540 is obviously a lot faster

This is what I am interested in. If you have numbers, please share.

andreas-bulling, 2023-09-24:
Never mind, found it. Go to the Victron integration, click on "Configure" and then "Rescan available devices. This will rescan all available devices". The old grid meter will not be removed but at least the new one will be added.

closed 2023-09-24 by andreas-bulling