sfstar/hass-victron #129: two pv-inverters available with different VRM instance, only one is shown

albundy0815, 2023-10-02:
I've two hoymiles pv-inverters implement in Victron. On the cerbo it's shown as well as in the VRM portal. Both have a unique ID, but only one of them is shown.
The unit 34 is visible but the second one with unit 35 is not scanned and not shown.
Debug Log:
2023-10-02 09:44:08.127 DEBUG (MainThread) [custom_components.victron.coordinator] Fetching victron data
2023-10-02 09:44:08.127 DEBUG (MainThread) [custom_components.victron.coordinator] {'20': ['pvinverter_registers'], '24': ['temperature_registers'], '25': ['temperature_registers'], '34': ['pvinverter_registers'], '40': ['gavazi_grid_registers'], '46': ['acload_registers'], '100': ['settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], '223': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '224': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '225': ['battery_registers', 'battery_detail_registers'], '226': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '227': ['vebus_registers']}
2023-10-02 09:44:08.204 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.078 seconds (success: True)
2023-10-02 09:44:12.017 DEBUG (MainThread) [custom_components.victron.config_flow] advanced options is set
2023-10-02 09:44:13.137 DEBUG (MainThread) [custom_components.victron.coordinator] Fetching victron data
2023-10-02 09:44:13.137 DEBUG (MainThread) [custom_components.victron.coordinator] {'20': ['pvinverter_registers'], '24': ['temperature_registers'], '25': ['temperature_registers'], '34': ['pvinverter_registers'], '40': ['gavazi_grid_registers'], '46': ['acload_registers'], '100': ['settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], '223': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '224': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '225': ['battery_registers', 'battery_detail_registers'], '226': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '227': ['vebus_registers']}
2023-10-02 09:44:13.279 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.142 seconds (success: True)
2023-10-02 09:44:18.127 DEBUG (MainThread) [custom_components.victron.coordinator] Fetching victron data
2023-10-02 09:44:18.127 DEBUG (MainThread) [custom_components.victron.coordinator] {'20': ['pvinverter_registers'], '24': ['temperature_registers'], '25': ['temperature_registers'], '34': ['pvinverter_registers'], '40': ['gavazi_grid_registers'], '46': ['acload_registers'], '100': ['settings_registers', 'settings_ess_registers', 'system_registers', 'system_battery_registers', 'system_dc_registers', 'system_charger_registers', 'system_power_registers', 'system_bus_registers'], '223': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '224': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '225': ['battery_registers', 'battery_detail_registers'], '226': ['solarcharger_registers', 'solarcharger_tracker_voltage_registers', 'solarcharger_tracker_registers'], '227': ['vebus_registers']}
2023-10-02 09:44:18.216 DEBUG (MainThread) [custom_components.victron.coordinator] Finished fetching victron data in 0.089 seconds (success: True)
2023-10-02 09:44:19.638 DEBUG (MainThread) [custom_components.victron.config_flow] host = 192.168.x.x
2023-10-02 09:44:19.638 DEBUG (MainThread) [custom_components.victron.config_flow] port = 502
2023-10-02 09:44:19.640 DEBUG (MainThread) [custom_components.victron.config_flow] connection was succesfull
....
2023-10-02 09:44:20.984 DEBUG (MainThread) [custom_components.victron.hub] no registers found for unit: 33
2023-10-02 09:44:20.985 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2600 count: 38
2023-10-02 09:44:20.989 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3 count: 91
2023-10-02 09:44:20.991 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 259 count: 68
2023-10-02 09:44:20.992 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 1282 count: 40
2023-10-02 09:44:20.993 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 771 count: 21
2023-10-02 09:44:20.994 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3700 count: 4
2023-10-02 09:44:20.996 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3708 count: 20
2023-10-02 09:44:20.998 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2048 count: 6
2023-10-02 09:44:20.999 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2307 count: 16
2023-10-02 09:44:21.000 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2700 count: 11
2023-10-02 09:44:21.000 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2800 count: 10
2023-10-02 09:44:21.001 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 2900 count: 4
2023-10-02 09:44:21.002 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3000 count: 8
2023-10-02 09:44:21.004 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3100 count: 3
2023-10-02 09:44:21.005 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3105 count: 2
2023-10-02 09:44:21.006 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3110 count: 8
2023-10-02 09:44:21.007 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3125 count: 4
2023-10-02 09:44:21.008 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3130 count: 9
2023-10-02 09:44:21.010 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3140 count: 4
2023-10-02 09:44:21.011 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3148 count: 21
2023-10-02 09:44:21.012 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3200 count: 24
2023-10-02 09:44:21.012 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3300 count: 9
2023-10-02 09:44:21.013 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3400 count: 4
2023-10-02 09:44:21.015 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3420 count: 5
2023-10-02 09:44:21.016 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3500 count: 10
2023-10-02 09:44:21.017 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3600 count: 4
2023-10-02 09:44:21.018 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3800 count: 1
2023-10-02 09:44:21.019 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3802 count: 26
2023-10-02 09:44:21.020 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 3900 count: 22
2023-10-02 09:44:21.021 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 4000 count: 12
2023-10-02 09:44:21.022 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 4100 count: 17
2023-10-02 09:44:21.023 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 4200 count: 12
2023-10-02 09:44:21.024 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 34 address: 4300 count: 12
[... cut ...]
2023-10-02 09:44:21.036 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 40 address: 259 count: 68
2023-10-02 09:44:21.037 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 40 address: 1282 count: 40
2023-10-02 09:44:21.038 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 40 address: 771 count: 21
2023-10-02 09:44:21.039 DEBUG (MainThread) [custom_components.victron.hub] result is error for unit: 40 address: 3700 count: 4

I'm having this issue on the latest version of hass-victron as well as on HA 2023-09
[image]

tziku, 2023-12-31:
Same problem here.
I have 1 MPPT Solar charger Unit ID: 224 and i add second one Unit ID: 215 and i can't add/see/edit is not existing in integration.

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar