sfstar/hass-victron #130: SOC = 0%

Drasek, 2023-10-05:
I got wrong values for the SoC in my installation:

[image]
[image]

If i have a look inside the device i see that the soc has the right value:
[image]

I can only match the charge/discharge data, not the soc - where does the system gets the soc from?
[image]

System:
Home Assistant 2023.10.0
Supervisor 2023.10.0
Operating System 10.5
Frontend 20231002.0 - latest

Victron:
Product | Cerbo GX
Firmware version | v3.10

Battery: Batrium LiFePo4 selfbuild system

Thank you very much for your help!

askpatrickw, 2023-10-09:
@Drasek can you post the YAML for your items. I'd like to see the full names of the sensors. I have had similar issues and I might be able to help find a pattern.

jumoog, 2023-11-12:
The `CCGX-Modbus-TCP-register-list-3.10.xlsx` states that these values are `Not supported by CAN.Bus BMS batteries.` If you are using a CAN.Bus BMS battery, you will always see the value 0.

angyp, 2024-03-05:
Guys the issue is when you enable write in the integration you have to select the "The DC voltage of your battery in V" and you only have the option of lifepo4_12v, lifepo4_24v, lifepo4_48v and selecting any of these removes the SOC probably because they are CAN bus types. But what if you have a 48V AGM battery?

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar