sfstar/hass-victron #131: More than one GX device not supportet

GerhardLang, 2023-10-10:
Hello.

I do have a configuration with more than one GX devices which I want to have in Home Assistant.
It would be great if the Integration would support multiple  instances.
Actually I need to create all modbus sensors  needed by hand. What a pity because the Integration is great.

pos-ei-don, 2023-10-10:
+1.
I also have 2 Systems, a 48V and a 24V System.

gris74, 2024-02-03:
+1 also have more than 1 system

cyon30, 2024-02-06:
any work around on this?

GerhardLang, 2024-02-07:
Hi, just my workaround:
I did create the modbus entries for the relevant values of each GX device, so I do not have such an amount of unnecessary values and I solved my problem with more devices.

cyon30, 2024-02-07:
I have 5 sites here at the main office and want some values at the main screen here at the ofiice

sfstar, 2024-02-25:
Further tracking of this feature request will be done in #162.
Please let your voice be heard over there.

closed 2024-02-25 by sfstar