sfstar/hass-victron #133: Unit of measure PvInverter

solaral, 2023-10-15:
The unit of measurement of the PV inverter is not correct. It is in KW by default, but should be in W. The problem is that if I change it to W, I get an incorrect value.
In KW, for example, it shows me 200 KW. If I change it to W, it makes 200,000 W. But the correct value would be 200W

bifbaf, 2023-10-17:
+1
Also notices the same problem

solaral, 2023-10-17:
Chnage the Line 557 in the const.py to 
`    "pvinverter_power_total": RegisterInfo(1052, INT32, UnitOfPower.WATT),`
After that the value is correct

bifbaf, 2023-10-17:
Where did you configure that?

sfstar, 2024-02-25:
Hello @solaral,

Thank you for opening this issue.
This is an specsheet mismatch between the value documented to be returned by the modbus server and the actual value being returned.
Currently I'm following up upon a support case with victron to get insight as to whether the reported value is correct and this integration needs to be patched or whether victron needs to patch the GX device software to report the values as they are documented.

sfstar, 2024-02-27:
V0.1.7 containing the fix for this issue has just been released.
Closing issue, feel free to re-open if issues persist after upgrading to v0.1.7

closed 2024-02-27 by sfstar