sfstar/hass-victron #136: Values for energy Dashboard get reseted

bifbaf, 2023-10-23:
Hi,

I set up a new sensor with the total PV consumption as a template 

`  - sensor:         
      - name: pv_production_total
        unique_id: "PVProductionTotal(kwh)"
        state_class: 'total_increasing'
        unit_of_measurement: 'kWh'
        device_class: 'energy'
        state: >
          {% set l1 = states('sensor.victron_pvinverter_l1_energy_forward_total_20') | float %}
          {% set l2 = states('sensor.victron_pvinverter_l2_energy_forward_total_20') | float %}
          {% set l3 = states('sensor.victron_pvinverter_l3_energy_forward_total_20') | float %}
          {{ (l1 + l2 + l3) | round(1, default=0) }}
`

This works fine and I get the sum of all three sensors into pv_production_total that I can use in energy dashboard.
For some reason once a day the statistic is reseted to the total amount and my static on the dashboard is wrong. I then remove the values manually and it is fixed.
This happens every day at the same time. Between 8-9 PM and 5-6AM.

[image]
[image]
[image]

bifbaf, 2023-11-08:
The Issue was the template.
This is fixed by adding this to the template 

`availability: >
          {{ states ('sensor.victron_pvinverter_l1_energy_forward_total_20')|float(0) != 0 and 
             states ('sensor.victron_pvinverter_l2_energy_forward_total_20')|float(0) != 0 and
             states ('sensor.victron_pvinverter_l3_energy_forward_total_20')|float(0) != 0 }}`

closed 2023-11-08 by bifbaf

pos-ei-don, 2023-11-08:
would using "has_value()" simplify this check ?

bifbaf, 2023-11-09:
I haven't tested it, maybe this will also work