sfstar/hass-victron #137: AC meter instance 35 not detected

apple047, 2023-10-25:
Hi, 
must thank you for your effort in building this integration, it makes life so much easier.
the issue I came across is one of the AC meters is not being identified, Modbus ID 35, the other 3 AC meters with ID 30 and 32 are identified correctly.
[image]
in the console, the meter is working as expected
[image]
from what I know ID 30-36 can be used for AC meters but couldn't verify this info 
[image]

sfstar, 2024-03-13:
Hello and thank you for opening this issue,

Unfortunately the id 35 isn't an ID that is scanned for currently.
When I initially wrote the victron integration these were the valid unit ids:

```
valid_unit_ids = [ 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 
                   11, 12, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 
                   31, 32, 33, 34, 40, 41, 42, 43, 44, 45, 46, 100, 223, 
                   224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 
                   235, 236, 237, 238, 239, 242, 243, 245, 246, 247
                   ]
```

I will open a ticket on the victron modbus issue tracker to get clarification regarding the list of actually valid modbus unit id's

sfstar, 2024-03-15:
Hello @apple047,

I've amended the list of valid unit ids on my end to accomodate installs where the devices started incrementally counting from 34.
With users reporting having 35 and 36 being assigned to additional device instances.
For new victron systems this should not be an issue any longer due to a new range of ids being specified in the xlsx.
Please follow the steps in https://github.com/sfstar/hass-victron/issues/182#issuecomment-1998254266 to install the version on main that should allow for you to discover the AC meter with instance id 35

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar