sfstar/hass-victron #138: Victron SolarCharge unavailable after HA update or just randomly

belmont, 2023-10-29:
Victron SolarCharge unavailable after HA update or just randomly. Reinstall of Victron integration solved it but reload did not. I am tired of reinstalling Victron every time, so i report it here

wernerhp, 2023-10-29:
```
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:885
First occurred: 21:17:07 (1 occurrences)
Last logged: 21:17:07

Updating state for sensor.victron_solarcharger_maxpower_today_223 (<class 'custom_components.victron.sensor.VictronSensor'>) took 1.105 seconds. Please create a bug report at https://github.com/sfstar/hass-victron/issues
```

wernerhp, 2023-10-31:
```
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/victron/sensor.py:157
Integration: victron (documentation, issues)
First occurred: October 30, 2023 at 05:18:40 (8 occurrences)
Last logged: 05:12:13

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 233, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 389, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 172, in async_update_listeners
    update_callback()
  File "/config/custom_components/victron/sensor.py", line 157, in _handle_coordinator_update
    self._attr_native_value = self.entity_type.decodeEnum(data).name.split("_DUPLICATE")[0]
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 712, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 1128, in __new__
    raise ve_exc
ValueError: 245.0 is not a valid solarcharger_state

```

belmont, 2023-11-03:
[image]
i updated the HA to latest and also reinstalled victron integration and it worked that time. However after a few hours i noticed that the solarcharge went "unavailable".

braamvh, 2023-11-05:
If it is a VE.Can charge controller, it might have changed ID from 0 to 100 - check if there's a new charge controller with the new ID.

belmont, 2023-11-06:
[image]
Yes it is VE.Can charge controller and it shows Model 0 when it is unavailable and when i reinstall Victron integration and the stats are back this page still showing model 0 for the solarcharger. Anyways thanks for the idea

marekchowaniok, 2023-11-25:
any updates ? it is happening to me as well

braamvh, 2023-11-26:
What I used to do is to just reboot HA once. The ID would change to 100, and stay there, unless the integration was re-installed and did a rescan. Then I changed all my automations and monitors to use this value, and not 0. Worked quite well for me.

Eventually I decided to change from VE.Can to VE.Direct, for this and other reasons. I only have 2 charge controllers in total.

belmont, 2023-12-18:
still the same issue with the latest HA. Note that the solarcharger ID is always zero, if you restart HA the solarcharger reporting stops working and as i wrote if i delete and reinstall Victron integration then it works again. The ID of solarcharger is still zero. VE.Can is modern and easy to use solution, i would not switch to VE.Direct.

braamvh, 2023-12-18:
Try this instead:
1) Restart HA, but do NOT re-install the HASS components
2) Check your devices. There will be 2 devices with ID 100, one showing with a lot of errors, the other working fine

The working one should contain all the entities for your solar charge controller, and you can use those without issues - they will stay at ID 100 after a reboot. Don't try to force it back to 0, rather just use the new value.

The issue is that ID 0 and ID 100 is the same for the GX device. This is because some modbus servers cannot access ID 0, so the CCGX also responds to ID 100 for these, this is documented in the modbus docs for Victron. But ID 100 is also used for the system values, so ID 100 causes a conflict that the integration can't handle properly.

Before I switched to VE.Direct I did it this way, and all my integrations and entities worked fine. I would have continued to use it this way, not switching to VE.Direct, but I found that the Cerbo GX history for the charge controllers only covers 1 day with VE.Can and 30 days with VE.Direct. Since I like having the history on the Cerbo, I switched to VE.Direct.

belmont, 2023-12-19:
this does not happen to me as you wrote. What happens are:
1) Restart HA, but do NOT re-install the HASS components
2) Check your devices. The solarcharger still have ID 0 just like before reboot and the Victron settings is ID 100 just like before reboot.

This is nothing to do with ID change because there is no ID change before or after, but i need to reinstall the Victron integration to make solarcharger reporting again

dummy74, 2024-01-05:
> this does not happen to me as you wrote. What happens are:
> 
>     1. Restart HA, but do NOT re-install the HASS components
> 
>     2. Check your devices. The solarcharger still have ID 0 just like before reboot and the Victron settings is ID 100 just like before reboot.
> 
> 
> This is nothing to do with ID change because there is no ID change before or after, but i need to reinstall the Victron integration to make solarcharger reporting again
> 
> **small good news:** with the latest HA and latest Victron integration, i dont need to reinstall the integration, just "configure" > "rescan" which is 3 second job, i just need to keep in mind doing it after every HA restart. For me 30 day history is not important, I keep VE.Can. This Victron integration should do a rescan automatically after every HA reboot, not hard to implement and we could close this ticket. For some reason if i check/enable rescan, this settings does not survive the reboot. ![Selection_954](https://private-user-images.githubusercontent.com/11472085/291502192-d76955c4-7213-4cca-b8a6-8a3181b4399c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0NDI2MTksIm5iZiI6MTcwNDQ0MjMxOSwicGF0aCI6Ii8xMTQ3MjA4NS8yOTE1MDIxOTItZDc2OTU1YzQtNzIxMy00Y2NhLWI4YTYtOGEzMTgxYjQzOTljLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA1VDA4MTE1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4ZTViYTkzMzg2NmNhMGE1ODM0Y2JjYzE1MjU2MWRhNGViZTAxNGFhZTg4YmMzNTJjOTI0NGI0YzA0ZTlmZmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fyJ2TVXii83j3TrprzMUkeyMe8vMzmWPxFOIy72-P8w)

Hello,

I have 2 solar charge controllers in my system and these are connected to the Cerbo via CAN. After each restart of HA, you have to perform a manual rescan, as the solar charge controller with ID 0 is assigned ID 100 after the restart.This is very annoying and is often only noticed later!

An automatic rescan after the restart would be very desirable. Perhaps this can also be implemented as an HA service. Then you can build your own automation?

bartsmetdv, 2024-01-06:
Having the same issue, an automatic rescan would indeed help a lot.

belmont, 2024-01-07:
lets keep this open until there will be better solution but at least this workaround does not take much time, we just need to remember to do after every restart

sanchosk, 2024-02-09:
I have the same issue with MPPT 250/100 - after every restart of HA the MPPT becomes unavailable, I have to go to rescan manually and then it works again. Would be great to have this automatically done during the startup of the integration, perhaps.

elianbgr, 2024-03-04:
> Имайки същия проблем, автоматичното повторно сканиране наистина би помогнало много.

+1

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar