sfstar/hass-victron #139: Victron value stops updating on UI, keeps getting value

sanchosk, 2023-10-30:
Not sure what happened, but for last few hours, my HA keeps displaying the value for my battery on the same level.
However, when I click on the details, it correctly shows the graph with the data, including the current value.
[image]
I did not do any update of HA or Victron that I am aware of.
Any idea what could be the reason this is happening?

sfstar, 2024-03-20:
Thank you for opening this issue,

Could you perhaps check the logs of the HA instance to see whether there are any warnings or errors present at the time the integration stops reporting the entities values?

sfstar, 2024-05-19:
Thank you for your reply and providing the debug logging.
I believe that the latest release v0.3.0 could also adress your issues since there were several stability improvements made between when this issue was reported and the latest release (released today).
Could you tell me if the issue is still occurring?

sanchosk, 2024-05-19:
Unfortunately, it behaves the same.
Installed 0.3.0, it required restart of HA and after restart I, again, have this:

-- Sent from /e/ Mail.

sanchosk, 2024-05-19:
Unfortunately, even after 0.3.0 update, it did not work.
What I did:
Update to 0.3.0
Restart HA
Wait 2 minutes
Check the MPPT
It was unavailable.
Rescan, again, did the trick and the data is back...
[image]

sanchosk, 2024-06-05:
Update - with the latest HA update, the restart now works OK - after reboot I have again the victron MPPT back.
Thanks!
Closing the issue.

closed 2024-06-05 by sanchosk