sfstar/hass-victron #140: Cerbo Reports Error on ReScan

askpatrickw, 2023-10-31:
Because of #47, frequent rescans are required, but then I see this in the Cerbo GX.

ERROR "Error processing function code 3, unit id 247, start address 865, quantity 2, src <HA IP>: Error finding service with device type system at device instance 256"

I think we're being too agreesive with the protocol. From looking at this.
https://www.victronenergy.com/live/ccgx:modbustcp_faq#error_codes_and_trouble_shooting

Code | Name | Description
-- | -- | --
0x03 | IllegalDataValue | The requested quantity of registers is invalid. See the modbus specs, http://www.modbus.org/specs.php, for the limits per function code.

Jonas001b, 2023-12-19:
Hi i have the same error did you fix it your self (what did you change? ) or is it still not working ?

askpatrickw, 2023-12-20:
Sorry, I'm no longer using this. Good luck!

sfstar, 2024-03-15:
Hello and thank you for submitting this issue,

Unfortunately these errors are to be expected upon scanning or rescanning for devices.
Since, before the scan, it is not known what unit id's and registers are populated.
The integration essentially goes to a list of all possible unit id and register set combinations in order to find which registers actually return data. 

This is a discovery function that is expected to trigger the invalid quantity warning at the victron end.
#47 is something that can be solved and is scheduled to be picked up in the coming period.
But for this issue in particulair I unfortunately have to close this issue as "intended behaviour".

closed 2024-03-15 by sfstar