sfstar/hass-victron #142: PV-Inverter with wrong unit for power entities

dummy74, 2023-11-05:
Hi,

Since today I have a Fronius Symo 6.0.3m on the AC out of my 3 MultiPlus 2 and was able to successfully register the Fronius inverter on my Cerbo. I then reconfigured the Victron addon in HA with the "Rescan" option. The Fronius is now displayed to me as a new device "pvinverter" with a total of 20 entities. Super!

Unfortunately, the unit is set incorrectly for the entities "victron_pvinverter_power_max_capacity_20" and "victron_pvinverter_power_max_capacity_20". The inverter has a max. power of 6000W, which is displayed as 6,000.0 kW.
[image]

bifbaf, 2023-11-08:
There is already a workaround for this check this issue may help you also

https://github.com/sfstar/hass-victron/issues/133

sfstar, 2024-02-27:
Change will be included in next release

sfstar, 2024-02-27:
V0.1.7 containing the fix for this issue has just been released.
Closing issue, feel free to re-open if issues persist after upgrading to v0.1.7

closed 2024-02-27 by sfstar