sfstar/hass-victron #143: sensor resetting to 0 on inverter disconnect

Daniel-Peeeh, 2023-11-08:
problem:
values of pv-inverters overall produced kwh from time to time reset to 0.
this happens 
  - randomly at night -> I think network connection between inverter (SMA) and cerbo is lost. reason unknown.
  - on restarts of the cerbo device

solution in Hass could be to create a filtered sensor. See solution here: https://community.home-assistant.io/t/how-to-prevent-entity-from-resetting-to-0/369363/4

for me it would be bad solution to create new sensors. i already have 1 year of data collected for the sensors created by this addon. removing them from energy dashboard is not an option.

is there a way to prevent the existing entities of my 3 inverters to reset? a filter to not set 0 values would be enough.

sfstar, 2024-05-19:
Thank you for opening this issue.
A few questions to better understand the problem:
- Does the VRM also show the 0 values reported by the SMA?
- Is your statistics data compromised?

Daniel-Peeeh, 2024-05-24:
Hey, thanks for response.
[image]
Screenshot is showing the SMA inverter sensor inside Victron connection last night. Value is going down to zero for a while and returning to old value afterwards. In Hass statistics this creates the 5900kwh production entry when returning to the higher value.

This is randomly happening to all my inverters (1x SMA, 2x Hoymiles, 1x Sungrow). Connection in Venus OS is done by dbus (Hoymiles, Sungrow) and native SMA connection. After this longer period of time, i can say that it is happining far more often to the SMA than the others. But I have no idea why. 

I do a correction of my statistics data everytime this happens. In the meantime HASS introduced the feature to search for the high statistics values, which makes it much more easy to find them. But still I would prefer a stable solution ;)

sfstar, 2024-07-02:
I get your struggles. Had the same issue with my goodwe inverter that randomly (daily) reset itself to 0 and immediatly after spiking back up to the previous value. Gets really annoying real fast (to fix the statistics data). I will look further into the issue to how we can prevent the statistics data from being contaminated. 
Do the affected inverters stay online during the entire night?

the78mole, 2024-10-05:
I have the same problems for almost a year and the problem (energy value peak) is increasing. I worked around that issue now by adding a template sensor and feeding this value to the energy dashboard. Unfortunately, I lost all historic data this way. I believe, returning an "unknown" state (or setting availability to false), when the energy value is zero, it is not a bad deal.

Here is my template sensor code, for people that want to also use a workaround:

```
template:
  - sensor
    - name: MiniPV Generation Total 
      icon: "mdi:counter"
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
      state: "{{ states('sensor.victron_pvinverter_l1_energy_forward_total_34') |float(0) }}"
      availability: "{{ states('sensor.victron_pvinverter_l1_energy_forward_total_34') | float(0) > 1 }}"
```

It is not perfect, but at least, it seems to avoid the (currently) 1000 kWh spikes in the energy dashboard. Let's see, what it gives during the next weeks.

closed 2025-03-05 by github-actions[bot]