sfstar/hass-victron #147: no values if battery on smartshunt is flat

pvtex, 2023-11-22:
i am using the newest version on the newest hass version.

i have  a smartshunt with an original ve.direct cable connected to my raspberry pi.

when the battery runs flat the whole addon stops working and i get no more values.
starting the charger i get values again.

there is perhaps a timeout problem with the smartshunt when there is insufficent battery power.

kwakerman, 2024-01-20:
Are you using a 12v system?  Smartshunt requires a minimum 6.5v to operate, if you are powering it from the battery and the battery is 'flat' as you say, is there at least 6.5v left in the battery to operate the Smartshunt?

sfstar, 2024-02-25:
Hello @pvtex,

Does the entire integration stop reporting values (while for example the vebus device / inverter is still running) or does only the smartshunt device stop reporting values.
Furthermore, It's important to know whether the GX device is still online once the battery goes flat.
Can you confirm this is the case?

pvtex, 2024-05-22:
The device is still online. Only smartshunt stops data

closed 2025-03-05 by github-actions[bot]