sfstar/hass-victron #157: Missing solarcharger

usekevin1988, 2024-01-20:
Hello everyone.

maybe someone has a idea.

My system:

1x Multiplus II 48/5000 
3x SmartSolar MPPT 100/20-48V 
1x Cerbo GX
1x Pylontech US5000
1x EM24 Ethernet

I cant find my third SmartSolar in HA.

I did a rescan, but nothing has changed.
[image]
[image]

any idea, what im doing wrong? Theyre all on the same way connected to my CerboGX

OlePZ9, 2024-01-22:
Sometimes the integration does not recognize MPPTs correctly , especially address 0/100. maybe there is something similar to higher addresses like 100.
You can include new sensors by using the Modbus integration by entering yaml code into configuration.yaml
[image]

OlePZ9, 2024-01-22:
[image]
This has to be entered before the other, take the ip address from your Cerbo GX

OlePZ9, 2024-01-22:
Use the mapping address for 280 instead of the number from my Codename take 215 instead of 226 for your third device

https://www.victronenergy.com/upload/documents/CCGX-Modbus-TCP-register-list-3.10.xlsx

[image]

OlePZ9, 2024-01-22:
Luckily the integration accepts manual Modbus sensors beside its own ones

sfstar, 2024-03-14:
Hello @usekevin1988,

Could you try the version on the main branch of this integration?
It contains the new unit id's and should allow for the solarcharger to be detected upon a rescan.

sfstar, 2024-03-29:
The release (v0.2.0) has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar