sfstar/hass-victron #159: Power total and max capacity

EckeFL, 2024-01-27:
The values for "Power max capacity" and "Power total" are to large by a factor of 1000.

Venus OS 3.11

[image]

sfstar, 2024-02-25:
Hello @EckeFL,

Thank you for opening this issue.
I've escalated my support request (that I opened last year may) with victron to get clarification on whether the documentation is leading and the victron modbus server needs to be patched or the software is leading and we can patch this on our end.
This will be picked up once victron replies.

sfstar, 2024-02-27:
Change will be included in next release

closed 2024-02-27 by sfstar