sfstar/hass-victron #160: The Variable ESS Maxdischargepower can only be set between 0 - 1800 W in GUI

Jochen754, 2024-01-27:
The Value can only be adjusted between 0 - 1800 W.

However if the parameter is limited in the Victron Web GUI itself, then it appears correct in de Homeassistant GUI.

However in this case it can only be adjusted downwards. Only after going under 1800 then it can also be increased again but with 1800 as the upper limit.

However for the Multiplus II Gx it should be able to go up to 2400 W.

Where is the limit of 1800 defined for ESS Maxdischargepower?

closed 2024-01-28 by Jochen754

Tomashaglund1, 2024-04-17:
@Jochen754 how did you resolve this? :)

[image]

[image]

[image]

Jochen754, 2024-04-17:
Hey,

nice that I can help you with this problem. The setting that needs to be changed at your end is in Homeassistant when setting up the integration:

In homeassistant go to:
- Settings
- Go to the overviet of the setup Devices
- Select the Victron Integration
- Select Configure
- Then set the Grid Voltagee and Phase setup correctly and also the following correctly:

"The AC (per phase) current limit of your grid in A"

For me 2400W with a 230V single phase system equals 10,43A so I set the apms to 12A.

Then save the settings. 

Now you should be able to set the limit up to 2400W or even larger (depends on the amp setting) however the Multiplus can only output max. 2400W even when set larger than that in the integration.

[image]

[image]

Tomashaglund1, 2024-04-18:
This is strange. Having threephase 230VAC / 25A on the mains to the house however even if that is set in the config it's not possible to rise above 1700W. Did restart HAOS after change was done also.

@sfstar any insight in what may causing this problem? On a Multiplus 2 48/3000

[image]

Jochen754, 2024-04-18:
@Tomashaglund1  
You need to raise the DC Limit from 35A to at least 50A to get the full 2400W output.

This DC current however needs to be supported by your battery.

In your screenshot it is still set at 35A redulting in a limit of:
35A x 48V = 1680W ~1700W shown

Tomashaglund1, 2024-04-18:
Ok clear.
305Ah cells here so there's margin to increase! Thanks I'll buy you a beer next time you're in Finland, Vasa :)