sfstar/hass-victron #161: VM-3P75CT Grid Meter - Connection lost after restart

21-Jailbraik-21, 2024-01-31:
VM-3P75CT Grid Meter - Connection lost after restart.

Hello, the new Grid Meter from Victron VM-3P75CT with the VRM number 0 no longer transmits after every restart of HomeAssistant.
After I rescan it is back.

Ask for help

21-Jailbraik-21, 2024-01-31:
Will the project not be continued here?
Or is it better to use MQTT?

mchiriciuc, 2024-02-18:
Hello,

I have the same problem with a SmartSolar MPPT RS 450/100.  It becomes unavailable after HA restart. After a restart it is back.

Hope you have the time to fix it. 

Thanks

21-Jailbraik-21, 2024-02-18:
https://gist.github.com/KidA001/dc5584db340cf3ad513f9415b20ed5fa

I'm trying to switch now because it's not working well - unfortunately.
I haven't finished it yet, but I can already access the data.

sfstar, 2024-03-15:
Hello and thank you for opening this issue,

This issue is related to #47.
It is currently planned that this error will be fixed in an upcoming release.

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar