sfstar/hass-victron #163: Values disappear every few hours

JBecker32, 2024-02-20:
I have had the victron integration in HA running since around a year now with absolutely no problems at all.
Now, some days ago, problems started with no change on the system (afair).

Values become unavailable after some hours running. The victron remote console (settings/services/Modbus TCP) shows error messages like :
ERROR "Error processing function code 3, unit id 100, start address 3500, quantity 10, src 192.168.0.26: Error finding service with device type generator at device instant 0"

HA log shows (every 10s):
2024-02-20 15:26:46.090 WARNING (MainThread) [custom_components.victron.coordinator] no valid data returned for entities of slave: 100 

As far as I understand this means that the HA (192.168.0.2) is requesting data from the Cerbo GX which cannot be resolved by the GX. Is this a configuration error? But how can I influence which data is read by HA? I do not find a configuration for this.

A reset of either the Cerbo or HA seems to fix this situation for a few hours and then it fails again.

JBecker32, 2024-02-22:
OK, solved the problem!
Reason was, as written before, the read access to unit id 100 address 3500. It seems that the victron GX closes down the TCP connection after 1 hour (?) of these permanent read errors.
I have now removed a line in the configuration of the hass-victron configuration in .storage/core.config_entries which says something like "system_generator_registers".
Removing this line stops the abovementioned read accesses to Victron GX and the system runs for hours now without any problem.
BR, Jörg.

sfstar, 2024-02-25:
Hello @JBecker32,

Thanks for opening this issue.
Could it be that your generator goes into some form of a deep sleep mode after x hours?
The integration polls all discovered register sets and if the discovery runs while the generator is awake this could explain the errors you are seeing after a few hours (when the generator is sleeping / unavailable again).
Hence, removing the generator register set resolving your stability issues.

JBecker32, 2024-02-25:
I do not have a generator, nor have I configured the GX to control one. So the GX is complaining that registers are read that do not exist (or are at least not populated).
This is how I understand it.
I can simulate the dame problem with a modbus master simulator requesting the same registers. Have not tested yet if it is one hour reading wrong values or a number of reads that makes the GX unresponsive.
BR, Jörg.

closed 2024-02-25 by JBecker32