sfstar/hass-victron #167: Invalid AC Power values for PVInverter after update to v0.1.6

dummy74, 2024-02-26:
I updated to version v0.1.6 yesterday. Since the update, the entities for my PV-Inverters
- sensor.victron_pvinverter_power_max_capacity_20
- sensor.victron_pvinverter_power_total_20

are displayed incorrectly. The values are too large by a factor of 1000!

[image]

sfstar, 2024-02-27:
I believe this behaviour didn't start from v0.1.6 but was already present and perhaps patched locally.
I'm running a support case by victron to get clarification as to where this can be more permanently fixed (either this integration or a GX device patch)

sfstar, 2024-02-27:
Change will be included in next release

sfstar, 2024-02-27:
V0.1.7 containing the fix for this issue has just been released.
Closing issue, feel free to re-open if issues persist after upgrading to v0.1.7

closed 2024-02-27 by sfstar