sfstar/hass-victron #172: Die Maßeinheitenw, W, W, W, W, W, W“ der Eingabesensoren aensor.victron_solarcharger_yield_power_239 sind nicht korrekt

pos-ei-don, 2024-02-28:
Hallo, 
Seit dem Update von HA auf 20124.02
bekomme ich von dem Addon diese Fehlermeldung:
Was kann ich tun?

Die Maßeinheiten sind nicht korrekt
Die Maßeinheiten „w, W, W, W, W, W, W“ der Eingabesensoren „sensor.tmpl_pv_fronius_frommodbus_power, sensor.victron_solarcharger_yield_power_239, sensor.victron_solarcharger_yield_power_235, sensor.victron_solarcharger_yield_power_231, sensor.victron_solarcharger_yield_power_238, sensor.pv9_hm1200_power, sensor.hmt_1800_power“ sind nicht kompatibel und können nicht mit der Geräteklasse „power“ der Sensorgruppe „sensor.grp_pv_sum_aktuell“ konvertiert werden.

Bitte korrigiere die Maßeinheit der Quellentitäten und lade den Gruppensensor neu, um dieses Problem zu beheben.

[image]

sfstar, 2024-02-28:
Hello and thank you for submitting and issue,

Please edit your question and translate it to english.
I'm able to read german, but neither german nor english are my native language.
By keeping english as the language of choice in this issue tracker the details of issues reported can be understood by a wider audience.

pos-ei-don, 2024-03-17:
Unit of measurements w, W, W, W, W, W, W of input sensors sensor.tmpl_pv_fronius_frommodbus_power, sensor.victron_solarcharger_yield_power_239, sensor.victron_solarcharger_yield_power_235, sensor.victron_solarcharger_yield_power_231, sensor.victron_solarcharger_yield_power_238, sensor.pv9_hm1200_power, sensor.hmt_1800_power are not compatible and can't be converted with the device class power of sensor group sensor.grp_pv_sum_aktuell.

Please correct the unit of measurements on the source entities and reload the group sensor to fix this issue.

pos-ei-don, 2024-03-31:
... It is still the case in New 0.2.0

sfstar, 2024-04-06:
Thank you for translating.
The integration uses the official constant UnitOfPower.WATT.
Therefore, I believe that the issue might be in the template sensor definition.
Could you share the yaml for that template sensor?