sfstar/hass-victron #175: include battery data in home-assistant energy-dashboard

Ekkib, 2024-03-03:
This would be very nice 😀 😃 😄 :

https://community.victronenergy.com/questions/266290/include-battery-data-in-home-assistant-energy-dash.html

braamvh, 2024-03-04:
I use sensor.victron_battery_history_chargedenergy and sensor.victron_battery_history_dischargedenergy and they work very well with the energy dashboard.

[image]

Ekkib, 2024-03-04:
Thanks ! I use exact the same variables, as the dropdown menu allows these values to be entered. But they are always zero, :-( . Sometimes "something" changes with new SW versions. But since 1 year no change. Always zero.

braamvh, 2024-03-04:
I think these are only available when you use a CAN managed battery. You can look through the other variables and see if there's another one that fits, or you will have to create your own. Bear in mind that the integration only shows the values returned via the Modbus registers, and the GX device does not expose everything this way. You can create a helper in the GUI to do the calculation for you, if needed.

sfstar, 2024-03-15:
@Ekkib Perhaps you can open an issue here: https://github.com/victronenergy/dbus_modbustcp/issues.
This is the issue tracker of the victron modbus server.
All values exposed by victron are documented and/or requested here.
Once the non can registers are exposed I can implement reading them in this integration.

Ekkib, 2024-03-29:
@sfstar Thanks for the idea, how to proceed. After browsing through many interesting information I still have no idea, where to start to support or to suggest an improvement. My initial post triggered the posting of some helpful hints. Unfortunately I could still not get, what I want. The HA-variables already exist, but only contain „0“ for me.

chrisguen, 2024-06-25:
> I use sensor.victron_battery_history_chargedenergy and sensor.victron_battery_history_dischargedenergy and they work very well with the energy dashboard. [...]

What type of battery/BMS are you using? I'm also using a battery connected via CAN bus, but my charged and discharged values are also 0. Are they reported by the BMS or calculated on the cerbo?

sergio-scl, 2024-06-30:
Same problem. 

sensor.victron_battery_history_chargedenergy and sensor.victron_battery_history_dischargedenergy always show 0.

My batteries are Pylontech US2000C.

sfstar, 2024-07-02:
Unfortunately this is a data provider issue. The modbus server does not return these registers with values other than 0 if a can-bus battery is used. In order to get better energy dashboard support victron needs to implement registers that provide the desired information. So that this integration might query it and make it available for users. Please see my previous comment on where you can submit the feature request with victron. (once implemented on their side I can add them to this integration)

ikke050, 2025-02-10:
did any of you solved this in the meantime, I have the same problem..

closed 2025-05-25 by github-actions[bot]

toxic049, 2026-02-15:
Any way to reopen this issue?