sfstar/hass-victron #176: Requires reboot of Cerbo GX after Homeassistant restart

AminShAT, 2024-03-03:
The integration always loses connection to the Victron installation, once you restart Home Assistant (e.g. for Updates). I always need to reboot the Cerbo GX afterwards to get it connected again.

braamvh, 2024-03-04:
This sounds like a network issue. The integration keeps working for me if either side is rebooted, but I do have my both my Cerbo and HA connected via a physical Ethernet cable (not to the same switch, but the switches are also connected).

AminShAT, 2024-03-04:
Same here. Even on the same switch. I don't think it is a network connection because the Node red connection works instantly after reboot. It is just the integration that can not talk to the Cerbo.

peregrines2, 2024-03-10:
Same Problem here unfortunately.
Using Ethernet ... so I think it is related to something else :(

AminShAT, 2024-03-10:
Something in HomeAssistant is completely screwing up my Modbus protocol. My Victron Wallbox lost connection to the Cerbo twice a day.

When I turned off HomeAssistant all together, it worked fine.

I have now HomeAssistant running without the Victron integration and currently also Nodered turned off.

I suspect the Victron integration being the root cause for this. I will keep you updated.

sfstar, 2024-03-13:
Thank you for opening this issue.
Would it be possible for you to enable debug logging for the integration and provide the logdump in this issue thread?
Thanks in advance.

sfstar, 2024-08-20:
Closing issue as stale.
Feel free to open a new issue if the issue persists.

closed 2024-08-20 by sfstar