sfstar/hass-victron #177: Interference of Victron-Integration with Victron EVCS?

AminShAT, 2024-03-03:
My EVCS loses connection to the Cerbo approx. Every 4-5 hours. The Modbus-TCP log in the Cerbo reports some strange error message with the IP address of the Homeassistant server. (see below).

Is the Victron-Integration the root cause for the problem?

[image]

Darrrren, 2024-03-10:
I have exactly the same issue without the victron integration. I think it's a poor wifi signal thats causing the problem, the charger is loosing connection with the wifi access point. Unfortunatley, the onboard wifi in these chargers is really bad!  Move/install a nearer wifi access point.

AminShAT, 2024-03-11:
It's actually not the WiFi, that was, when my Solateur thought as well, so I put a repeater just next to the EVCS, but that did not help.

It is actually some software on my HomeAssitant that messes up the Modbus. I suspect the Victron Custom Integration. I have it currently switched off and everything works just fine. I also removed the repeater again.

I will try to switch on the suspected components in HA one by one now to see which one causes the trouble.

sfstar, 2024-03-13:
Thank you for opening this issue.
The error reported in the victron GX UI references that the device was polled by this integration which unfortunately the GX device was unable to do.

I get the same errors on my end when I disconnect my carlo gavazzi grid meter from the GX device.
It indicates the following:
HA -> GX -X> Device being polled

Where the -X> indicates the location of communication failure.
If you where to implemented your own modbus tcp client you would generate the exact same error when requesting a register from a device that is temporarily unavailable to be polled.

AminShAT, 2024-03-13:
Hi, thanks for your detailed answer. Currently it is actually running fine, let me explain what I did and what might be the source of issue.

Also my other reported issue #176 is gone for now, they seem to be related.

I had constantly losing connections between my EVCS and the Cerbo, sometimes with this error message, sometimes without.

So I switched off Homeassistant completely for a couple of days and all components in the Victron ecosystem worked just fine.

Then I switched on Homeassistant again but disabled the Victron integration and Nodered. Again, everything works just fine.

Then I switched on the Victron integration and for the first time it worked immediately (without restarting the Cerbo) and up until now everything is still working fine.

This evening I will switch on Nodered again, where I am using the Victron addon and I will see, if the problem occurs again.

#My gut feeling right now is, that with the Victron addon in Nodered activated on the Homeassistant PI and the Homeassistant Victron integration running in parallel, they mess up Modbus somehow.

If this is the rootcause it would just mean, you can not run the integration and Nodered workflows in parallel on the HA, but you can still run Nodered workflows on the Cerbo's Nodered (what is currently running at my installation).

AminShAT, 2024-03-14:
There is definitely an incompatibility with running Nodered workflows that also access the Cerbo. I had nodered switched on again for testing and the integration failed to reload. Only after I restarted the Cerbo. Same behaviour as previously.

closed 2025-03-05 by github-actions[bot]