sfstar/hass-victron #178: ValueError: 245.0 is not a valid solarcharger_state

wernerhp, 2024-03-04:
```
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/victron/sensor.py:157
Integration: victron (documentation, issues)
First occurred: 05:48:54 (5 occurrences)
Last logged: 05:48:58

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 256, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 412, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 183, in async_update_listeners
    update_callback()
  File "/config/custom_components/victron/sensor.py", line 157, in _handle_coordinator_update
    self._attr_native_value = self.entity_type.decodeEnum(data).name.split("_DUPLICATE")[0]
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 744, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/enum.py", line 1158, in __new__
    raise ve_exc
ValueError: 245.0 is not a valid solarcharger_state

```

sfstar, 2024-03-13:
Hello and thank you for opening this issue,

It seems that there are more solarcharger states than documented by victron.
Since 245 isn't specified in the documentation.
Were you able to see in the GX UI what the charger was doing at the time of this error? (Might help to deduce what the solarcharger_state 245 entails).

wernerhp, 2024-03-13:
`"254": "Off"`,
[Source](https://github.com/victronenergy/node-red-contrib-victron/blob/30d49be6d27a4c172aef589ed335111b818d8e54/src/services/services.json#L4402)

sfstar, 2024-03-17:
Thank you for the information.
However according to this PR: https://github.com/sfstar/hass-victron/pull/150
245 could also reference a starting state.
Since 0 is allready defined as "off" this seems more likely to me.
Could you confirm that 245 is only reported right before the solarcharger starts?

wernerhp, 2024-03-17:
Ah, ok, here's the doc they referenced
[image]
https://www.victronenergy.com/upload/documents/BlueSolar-HEX-protocol.pdf

sfstar, 2024-03-30:
@wernerhp thanks for the resource.
Will add state 245 as "WAKE-UP" state for now.

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar