sfstar/hass-victron #179: missing Solarcharger after change Charger hardware

Heiko00000, 2024-03-05:
Hello,

I build a HA System with RP5 and actuell software Januar 2024. 
Yesterday, i musst change  the MPPT 150/60 (faulty) against the same type.

The Rescan find no Solarcharger.
Delate the Victron Hub in HA and Scan, 5 Items find, no Solarcharger.

New Apdate  for Victron Cerbo GX von 3.13 to 3.22, no improvement.
New Apdate HA to Home Assistant (2024.2.3) to 2024.2.5. no improvement.
New apdate hass-victron to V0.1.7, no improvement.

Solarcharger have the Number 216 /281

Thanks Heiko

Heiko00000, 2024-03-07:
Hello,

I did a test, reconnected the defective solar charger to the Cerbo and after another scan HA found it. HA model number 224, Victron ID 278.
Reconnected the new solar charger, HA cannot find a charger.
Disabled the defective and disconnected solar charger in HA, no improvement.
[image]

Does anyone have an idea where the error is?

Thank you, Heiko

Heiko00000, 2024-03-13:
Hello und thanks for your help.

The Solarcharge have the VRM Instance 281.
In Hass-Victron it should be 216.

Heiko

sfstar, 2024-03-13:
Hello @Heiko00000,

Is either 281 or 216 the id that is visible on this screen:
[image]
related to the victron solarcharger?

sfstar, 2024-03-13:
You can try installing the latest main version by:
Going to hacs
Selecting redownload for the integration.
There you select "main" as the version (since the unit id change hasn't been released yet).

After downloading, restart HA and perform a rescan of all available devices.
This should then yield the new solarcharger as an device.

Heiko00000, 2024-03-16:
Hello, 
the solarcharger is found and works, thank you.  

But I had to enter the number 216 directly in the "const.py" because the call "main" kept getting replaced by 0.1.7.  

Greetings Heiko

sfstar, 2024-03-29:
The release has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar