sfstar/hass-victron #180: OpenDTU PV Inverter missing

CerboJoe, 2024-03-12:
2 of 3 Virtual Inverters are not found. the third , exactly the same, ist found and works well. So ID 32 is online, 33 and 34 are missing.
Greetings and thank U so mutch for the work,
Joe

sfstar, 2024-03-13:
Hello and thank you for opening this issue.
A few questions:
- The modbus id's 33 and 34 aren't being detected while 32 is?
- Are the inverters online and reporting values in the GX device UI at the time the integration is scanning for registers?
- Are you able to create a modbus integration configuration to read one of the values of the PV inverters with slave address 33 and 34? (This should confirm whether the pv inverter is actually exposed).

sfstar, 2024-03-15:
Hello @CerboJoe,

Could you perhaps install the latest main version by following these instructions:
https://github.com/sfstar/hass-victron/issues/182#issuecomment-1998254266

After which please reconfigure the integration to rescan all available devices (when all inverters are online and reporting in the GX device).

CerboJoe, 2024-03-16:
I'm sorry.It dosent change anything. even not, after a full reinstallation. The HM-300 is shown, the two HM-1500 are not.

sfstar, 2024-04-19:
Thanks for checking.
Could you perhaps share the entries located under:
"settings -> services -> modbus/TCP -> available services" on your GX device?

sfstar, 2024-04-19:
Also please add the following configuration to your configuration.yaml file to further troubleshoot whether the inverters values are actually retrievable from the gx device:
```
modbus:
  - name: victron
    type: tcp
    host: <ip of the gx device>
    port: 502
    retry_on_empty: true
    retries: 10
    close_comm_on_error: true
     sensors:
       - name: pv_inverter_l1_voltage
         unit_of_measurement: "V"
         slave: 33
         address: 1027
         device_class: voltage
         state_class: measurement
         data_type: uint16
         scale: 0.1
         precision: 2
         offset: 0
```
Take note to customize the ``` <ip of the gx device> ``` part to the ip of your GX device.

After adding the configuration please restart your home assistant instance and check whether an entity is created with the name ``` pv_inverter_l1_voltage ``` and it's state is being populated with the inverters voltage values that are also visible in the GX device.
Please report your findings so that this issue can be further troubleshooted.

sfstar, 2024-08-20:
Closing issue as stale.
Feel free to open a new issue if the issue persists.

closed 2024-08-20 by sfstar