sfstar/hass-victron #183: Feature request: set charge mode of EVCS

AminShAT, 2024-03-14:
Are you planning to implement to be able to set the charge mode of the EVCS from within your integration?

Usually I charge via the auto mode, but sometimes it would be necessary to switch to manual mode in order to charge from grid.

OlePZ9, 2024-03-15:
Already there but wrong naming implemented
[image]

OlePZ9, 2024-03-15:
You’ve got the right register, but wrong naming. Register used for position (AC1in…) ist the register for setting the mode. You only need to rename it. I did it once manually, but after the last update it was gone

OlePZ9, 2024-03-15:
[image]

closed 2025-03-05 by github-actions[bot]