sfstar/hass-victron #192: Missing Victron Solarsmart MPPT

johnh1917, 2024-03-27:
Hi everyone,

I have an issue with my system, where one out of two Solarsmart MPPTs is not detected.

I have following system:

1x Multiplus II 48/3000
2x SmartSolar MPPT 100/20-48V
1x Cerbo GX
1x Victron Smartshunt
3x water level sensors
1x temp sensor

Every other device has been correctly detected expect one MPPT.

I have done few rescan, restarted the Cerbo but nothing has changed.

If I view the available ModBus TCP services on Cerbo GX, the charger is listed, unit ID 215.

Any help would be much appreciated.

Thanks,

sfstar, 2024-03-29:
Hello @johnh1917,

Could you perhaps install the main version of this application.
I contains the patch for the issues you are facing. (Will be released soon)

sfstar, 2024-03-29:
Duplicate of #157

sfstar, 2024-03-29:
The release has been shipped.
Closing issue, feel free to re-open if the issue persists with the new release

closed 2024-03-29 by sfstar