sfstar/hass-victron #200: Please include addition ModBUS IDs for inverters

AminShAT, 2024-04-06:
As discussed please allow IDs above 36 for AC inverters.

a-wolter, 2024-04-11:
Just added some opendtu inverters with id 34 and 35 and they also don’t show up. I think it’s maybe not only is 36 but I will add a third opendtu inverter soon with id36
I already redownloaded the integration set to “main” version. And restarted homeassistant. 

In the system settings the slider is set to add new devices. But only inverter ID 20 is there. 34 and 35 are still missing. 

a workaround could be to change the id in the opendtu config so that it will show up in the cerbo with a different id and the integration van pull it. But what would be the correct ID range for additional pv inverters?

sfstar, 2024-04-19:
@a-wolter please add your comment to #180 so that the opendtu related issues can be tracked in a single place.

sfstar, 2024-05-19:
The issue has been shipped in the latest release (v0.3.0) closing issue as resolved.
Feel free to re-open if the issue persists with the v0.3.0 release

closed 2024-05-19 by sfstar