sfstar/hass-victron #214: Cannot change IP address

owenashurst, 2024-06-06:
If the Cerbo GX changes IP address (I have it temporarily setup using ethernet over WiFi), then you cannot update the IP address when reconfiguring. You have to delete it, and re-add it using the new IP address. 

Tl;Dr It keeps failing on "failed to connect" whenever I try submitting the new IP via reconfigure.

Apfuntimes, 2024-08-05:
Confirming that this is an issue, and worse. With 300+ entities across 5 devices, anytime I have to delete and re-add it means going through and disabling or hiding 100’s of entities. :-(

JohnMcLear, 2024-12-20:
STR:
Add device
Finish adding
Go back to Settings > Devices > Victron
Click three dots
[image]
Click Reconfigure
Insert New IP Address
Click Submit
Will error as
[image]

Even if IP/hostname/port is correct and working.

peetjvv, 2025-01-04:
@JohnMcLear , out of curiousity, what does STR stand for?

I have tried the steps outlined by JohnMcLear and ran into the same issue. I've updated to v0.3.0 of the integration.

peetjvv, 2025-02-18:
Judging by the release logs it doesn't look like this issue has been addressed yet. However, I figured out a manual process to change the host / IP address. Proceed with caution though as you will be manually editing the core config file for your entire setup.

1. Create and download a backup!
2. Navigate to `\\homeassistant\config\.storage` (where `\\homeassistant` is your hostname). I used the "Terminal & SSH" add-on. Note that `.storage` is a hidden folder.
3. Edit the `\\homeassistant\config\.storage\core.config_entries` file.
4. Carefully look for the victron integration.
5. Edit the value for `host` to your desired value.
6. Save the changes.
7. Restart
8. The victron integration should see your GX device again and pull through some data.

Ives25, 2025-03-22:
Hi
You need to delete and then add a new bridge using the new IP address.

closed 2025-06-25 by github-actions[bot]